MiCode/Xiaomi_Kernel_OpenSource #2: How to compile

rahulrdhobi, 2015-01-09:
Please provide compilation steps

sheikhrr, 2015-01-16:
my setup for succesfully compilation-- install ubuntu 14.10. install Orcal java6, all essential libraries (google it please) and linaro toolchain arm-linux-gnueabi-linaro_4.7.4-2014.06-build_2014_09_01.tar.xz  and armani_user_defconfig

rahulrdhobi, 2015-01-16:
Ok have compiled successfully but after flashing new boot.img by fastboot my device it keep rebooting

Redmaner, 2015-01-16:
Are you getting passed the kernel boot screen?
You need initramfs and all the specific commands to get a working boot.img

rahulrdhobi, 2015-01-19:
what I have done is got boot.img from redmi 1s and used used unpack-repack tool from http://forum.xda-developers.com/redmi-1s/general/guide-unpack-repack-kernel-t2908458 , I have replaced compiled zImage with unpacked boot.img and repacked boot.img after that flashed new boot.img with fastboot but phone gets rebooting after MI logo

TheStrix, 2015-01-19:
@rahulrdhobi, where are you getting problem? What's the size of dtb.img you getting?

rahulrdhobi, 2015-01-19:
@TheStrix : I have unpacked original boot.img where I got initrd,dt_image,kernel and ramdisk.gz size of dt_image is around  5.6 MB , what I have done was copied compiled zImage into unpacked boot.img  removed kernel  renamed zImage to kernel and repacked it flashed new boot.img through fastboot. is there any step which I'm missing

TheStrix, 2015-01-19:
Compile dtb.img too from the source.. See command for compiling dtb.img from here:https://github.com/xiaolu/mkbootimg_tools 
And use dtbTool from here: https://github.com/LuffyXDA/BeastKernel_Armani/blob/BeastKernel/dtbTool

rahulrdhobi, 2015-01-19:
@TheStrix: Thanks for quick response will try and let you know results :) :+1:

rahulrdhobi, 2015-01-24:
@TheStrix:I have compile dtb.img from source using dbtTool but same still issue do I need to flash boot.img with compiled drivers also?

rahulrdhobi, 2015-01-24:
Here are logs for kernel http://pastebin.com/qBH1hViJ and adb logcat http://pastebin.com/f2yS5ieC
my phone is in bootloop

rahulrdhobi, 2015-02-03:
someone told me that this kernel wont work with stock MIUI rom its only works with custom rom like cyanogenmod 11. Can anyone know reason?

mrafra, 2015-02-03:
Anybody luck with Dior?

mrafra, 2015-02-04:
Same isue, bootloop...

luntik2012, 2016-07-23:
Provide build instructions