MiCode/Xiaomi_Kernel_OpenSource #3: Please Correct these lines !  starters are facing problem because of these errors!

sheikhrr, 2015-01-17:
Errors are-:  

arch/arm/mach-msm/smd_init_dt.c:24:25: fatal error: smd_private.h: No such file or directory

compilation terminated.

SOLUTION:- 

arch/arm/mach-msm/smd_init_dt.c:24:25   
from
 <smd_private.h>     to    "smd_private.h" 

sound/soc/msm/msm8226.c:30:40
from            
 <qdsp6v2/msm-pcm-routing-v2.h>    to    "qdsp6v2/msm-pcm-routing-v2.h"

drivers/misc/tspdrv/tspdrv.c:34:20:
from
<tspdrv.h>   to    "tspdrv.h"

drivers/misc/tspdrv/tspdrv.c:43:24:
from
<ImmVibeSPI.c>    to    "ImmVibeSPI.c"

sound/soc/msm/qdsp6v2/rtac.c:28:21:
from 
<q6voice.h>   to     "q6voice.h"

drivers/misc/tspdrv/tspdrv.c:86:41:
from
<VibeOSKernelLinuxHRTime.c>   to   "VibeOSKernelLinuxHRTime.c"

drivers/misc/tspdrv/tspdrv.c:84:37:
from
<tspdrvOutputDataHandler.c>    to     "tspdrvOutputDataHandler.c"

balika011, 2015-01-31:
This is not a problem, just they building the source worng way...

Manu4156, 2015-01-31:
reply@reply.github.com

mrafra, 2015-02-02:
Same problem...
I am new in this, can you someon pls explain me, how to compile it right way?
How to use generic unaligned.h?

M1cha, 2015-02-02:
@mrafra You're doing nothing wrong. Qualcomm kernel just need these fixes when compiling outside the AOSP tree. just apply them and u're fine.

But these patches don't need to be in the upstream kernel.

mrafra, 2015-02-02:
Ok, fixed, but I have one more error:
drivers/mfd/lm3533-core.c:626:5: warning: 'pdata' may be used uninitialized in this function [-Wuninitialized]
with dior_user_defconfig and NDK/toolchains/arm-linux-androideabi-4.6

M1cha, 2015-02-02:
@mrafra  struct lm3533_platform_data *pdata = NULL;

mrafra, 2015-02-03:
Next errors:

In file included from drivers/misc/tspdrv/tspdrv.c:52:
drivers/misc/tspdrv/ImmVibeSPI.c: In function 'ImmVibeSPI_ForceOut_SetSamples':
drivers/misc/tspdrv/ImmVibeSPI.c:433: error: implicit declaration of function 'KBUILD_STR'
drivers/misc/tspdrv/ImmVibeSPI.c:433: error: 'tspdrv' undeclared (first use in this function)
drivers/misc/tspdrv/ImmVibeSPI.c:433: error: (Each undeclared identifier is reported only once
drivers/misc/tspdrv/ImmVibeSPI.c:433: error: for each function it appears in.)

drivers/media/platform/msm/vidc/msm_venc.c: In function 'try_set_ctrl':
drivers/media/platform/msm/vidc/msm_venc.c:1372: error: unknown field 'val' specified in initializer
drivers/media/platform/msm/vidc/msm_venc.c:1446: error: unknown field 'val' specified in initializer

sheikhrr, 2015-02-04:
@mrafra... Which GCC you are using.... Please use linaro 4.9 or 4.7

mrafra, 2015-02-04:
I can't find prebuilt Linaro toolchain 4.9 or 4.7

mrafra, 2015-02-04:
Ok, compiled successfully after some changes in code:

drivers/misc/tspdrv/ImmVibeSPI.c:433:5:
pr_info("%s: Only support single sample for ERM\n",__func__);

drivers/mfd/lm3533-core.c:626:5:
struct lm3533_platform_data *pdata = lm3533->dev->platform_data;

NDK/toolchains/arm-linux-androideabi-4.6 works fine...