MiCode/Xiaomi_Kernel_OpenSource #4: Request release of 紅米 1's Linux© kernel source code

Vdragon, 2015-01-25:
Please comply to the General Public License v2 and release 紅米 1's kernel source code.

qiansen1386, 2015-03-31:
I thought it has been done?
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/armani-jb-oss

Vdragon, 2015-03-31:
@qiansen1386 As far as I can see HM1S is not same to HM1(the very, original model)

qiansen1386, 2015-04-01:
I see...

sidealice, 2015-04-17:
紅米 2 呢?

Vdragon, 2015-04-17:
@sidealice 請另外建檔新的 issue，謝謝

Vdragon, 2015-04-26:
Hi everyone concerning with this issue, please contribute other languages title(& issue summary also) so we can get more attention from @MiCode ;)

zwliew, 2015-05-24:
This will pretty much never happen considering Mediatek's track record.

git-rbanerjee, 2015-10-11:
Redmi Note 3g's sources are still not released, 
Xiaomi is hopeless.

parisqian-misa, 2015-10-12:
What if the kernel is not even modified?

Vdragon, 2015-10-12:
@parisqian-misa 

> What if the kernel is not even modified?

Whether the kernel source is modified or not according to GPLv2 source code still need to be provided with it's binary product.

nick37332001, 2015-10-13:
I guess if other companies will release kernel source *cough* *cough* *xiaomi* MediaTek may follow and release the kernel as well.

qiansen1386, 2015-10-13:
@Vdragon  I found that Redmi 1 is using MTK processor, and I heard about rumors about MediaTek do held the source code of their kernels. Perhaps, just maybe, the decision of not releasing kernel source code actually comes from some secret biz deal between Xiaomi and MediaTek.

I am not saying that Xiaomi is innocent, they are definitely wrong, but in their management's perspective this device may not be important anymore.
Since many Mainland Chinese do not have enough legal awareness (Thanks to the great law-fucking party and information-blocking GFW), also Redmi 1 is almost a mainland-China-Only device, plus it has released for 2 years. I can see the chances of releasing the code is pretty low, the focus of the company must be shifting to their next big release, due to the fierce market competition.

zwliew, 2015-10-13:
@qiansen1386 Any Mediatek device will almost never get an open sourced kernel. Any other Xiaomi device will probably have to wait at least a year to get one.

nick37332001, 2015-10-13:
@zwliew what about the mi4i? There must be some reason why they release it so fast. Oh yah... It's because Mi India developed the device.

parisqian-misa, 2015-10-13:
@nick37332001  Good point... :joy:

zwliew, 2015-10-13:
@nick37332001 I'm pretty sure it has more to do with it being an international device compared to one that is more focused towards the Chinese and other Asian markets.

nick37332001, 2015-10-13:
@zwliew More likely my point since the snapdragon 615 was also developed by Qualcomm india

further3344, 2016-02-20:
有其他的同cpu的手機已經放出代碼了