MiCode/Xiaomi_Kernel_OpenSource #9: Missing WLAN driver source code

kn8wolf, 2015-03-24:
Hi,

Where do I find the source code for the Pronto WLAN driver of Prima specific to REDMI Note 4G?

It is required to successfully compile the kernel and make use of the WiFi on the handset. 

This kernel source is missing the driver's source code and hence wifi will not get enabled in custom kernel after compilation.
Other devices' kernel sources do provide the drivers for their h/w (pronto wlan) for compilation.

It would be great to have for DIOR as well..
CONFIG_PRONTO_WLAN=m

fergy, 2015-04-24:
https://android.googlesource.com/kernel/msm/+/142f315b1fac51d07b9693c68a202c3a0211d965/drivers/staging/prima

liuguo09, 2015-05-27:
Please refer to this [FAQ](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/wiki/FAQ#where-is-wifi-driver-module)

fergy, 2015-05-27:
I got prima sources as those are available on github

Vdragon, 2015-06-21:
@kn8wolf Since this issue is solve could you close it?

closed 2015-08-09 by kn8wolf