MiCode/Xiaomi_Kernel_OpenSource #10: Missing armani-kk-oss branch.

balika011, 2015-03-29:
KK armani kernel source is missing.

TheStrix, 2015-03-31:
Yeah!! Xiaomi, please publish it ASAP!

balika011, 2015-04-11:
Its still missing!

tarunkapadia93, 2015-06-15:
bump still missing -_-

rooque, 2015-06-16:
Check this https://github.com/SlimRoms/kernel_xiaomi_armani

tarunkapadia93, 2015-06-16:
that kernel is good but has a vidc bug where in you cant save videos of any kind tried to fix it but was enable to do so rebelos himself tried n so did i

Vynikal, 2015-07-01:
And we still wait...

TheStrix, 2015-07-06:
So, MIUI V6 based on Kk is now officially released. Waiting for Xiaomi to to release Kk kernel source for armani. 
Hope good :wink:

balika011, 2015-07-06:
and also for other xiaomi devices....

tarunkapadia93, 2015-07-07:
KK sources for armani pls

liuguo09, 2015-07-08:
Armani KK is merged into Dior long time ago
Please refer to https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/dior-kk-oss
And use lunch to build armani

Vynikal, 2015-07-08:
@liuguo09 O.O really??
There's no defconfig for armani as I look..

balika011, 2015-07-08:
@liuguo09 Why are you lining?

Vynikal, 2015-07-14:
@liuguo09 If is it so, please tell me how to compile it and which defconfig should I use.

liuguo09, 2015-07-17:
Really? Sorry i didn't verify the KK branch. sorry, I will do a check

Vynikal, 2015-07-23:
@liuguo09 so what!?

Vynikal, 2015-08-05:
-_-

rebelos, 2015-09-30:
@liuguo09 we are almost down to a MIUI7 release it would be great if you can push it, since dior-kk-oss has a lot of things missing

TheStrix, 2015-10-10:
Bump! @liuguo09 made armani-kk-oss branch! 
Updated sources incoming!!!!!!!!

LuffyXDA, 2015-10-10:
finally source is up!!!!!!

TheStrix, 2015-10-10:
They uploaded armani-kk-oss now! Check! :smile:

LuffyXDA, 2015-10-10:
@TheStrix Yeah now merge with dior-kk-oss
lets try ;)