MiCode/Xiaomi_Kernel_OpenSource #11: Error in building the kernel

arnabbiswasalsodeep, 2015-04-02:
>arch/arm/mach-msm/smd_init_dt.c:23:25: fatal error: smd_private.h: No such file or directory
>#include smd_private.h
>                         ^
>compilation terminated.
>make[1]: *** [arch/arm/mach-msm/smd_init_dt.o] Error 1
>make: *** [arch/arm/mach-msm] Error 2
>make: *** Waiting for unfinished jobs....

Error while building the kernel in the file (line 24)
>arch/arm/mach-msm/smd_init_dt.c
cannot find smd_private.h

Solution:-
Replace <> with ""

emceethemouth, 2015-04-17:
Thats not the proper way to fix this, you should do this: https://github.com/emceethemouth/kernel_cancro/commit/f5dcfd29b029deeafacb39b82bd5ff94cb437fee

arnabbiswasalsodeep, 2015-04-18:
@emceethemouth Thanks a lot
what are the fixes you included in it?

closed 2015-10-13 by arnabbiswasalsodeep