MiCode/Xiaomi_Kernel_OpenSource #14: Compilation Error [CANCRO]

Xephi, 2015-04-02:
Again an another error : 

In file included from drivers/tspdrv/tspdrv.c:54:0:
drivers/tspdrv/ImmVibeSPI.c: In function 'drv2604_write_reg_val':
drivers/tspdrv/ImmVibeSPI.c:681:4: error: implicit declaration of function 'KBUILD_STR' [-Werror=implicit-function-declaration]
drivers/tspdrv/ImmVibeSPI.c:681:4: error: 'tspdrv' undeclared (first use in this function)
drivers/tspdrv/ImmVibeSPI.c:681:4: note: each undeclared identifier is reported only once for each function it appears in
drivers/tspdrv/ImmVibeSPI.c: In function 'drv2604_read_reg':
drivers/tspdrv/ImmVibeSPI.c:732:2: error: 'tspdrv' undeclared (first use in this function)
cc1: some warnings being treated as errors

scripts/Makefile.build:307: recipe for target 'drivers/tspdrv/tspdrv.o' failed
make[2]: *** [drivers/tspdrv/tspdrv.o] Error 1
scripts/Makefile.build:443: recipe for target 'drivers/tspdrv' failed
make[1]: *** [drivers/tspdrv] Error 2
Makefile:950: recipe for target 'drivers' failed
make: *** [drivers] Error 2

rooque, 2015-04-02:
Compiling with what GCC ?

Xephi, 2015-04-02:
https://android.googlesource.com/platform/prebuilts/gcc/linux-x86/arm/arm-eabi-4.6

rooque, 2015-04-02:
Try with 4.7

Xephi, 2015-04-02:
Same error :/

DempaPower, 2015-04-03:
Xiaomi; it's great to give a kernel, but you could give instruction on how to build too. And which dependancy are necessary. My next phone will be another one I think.

stefant234, 2015-04-03:
Try GCC 4.9.3 (http://forum.xda-developers.com/showthread.php?p=36677987#post36677987)
I haven't tried it yet though, have more important work than this mess here

Xephi, 2015-04-03:
Again, same error at the same point :

In file included from drivers/tspdrv/tspdrv.c:58:0:
drivers/tspdrv/ImmVibeSPI.c: In function 'drv2604_write_reg_val':
drivers/tspdrv/ImmVibeSPI.c:681:4: error: implicit declaration of function 'KBUILD_STR' [-Werror=implicit-function-declaration]
drivers/tspdrv/ImmVibeSPI.c:681:4: error: 'tspdrv' undeclared (first use in this function)
drivers/tspdrv/ImmVibeSPI.c:681:4: note: each undeclared identifier is reported only once for each function it appears in
drivers/tspdrv/ImmVibeSPI.c: In function 'drv2604_read_reg':
drivers/tspdrv/ImmVibeSPI.c:732:2: error: 'tspdrv' undeclared (first use in this function)
cc1: some warnings being treated as errors
scripts/Makefile.build:307: recipe for target 'drivers/tspdrv/tspdrv.o' failed
make[2]: *** [drivers/tspdrv/tspdrv.o] Error 1
scripts/Makefile.build:443: recipe for target 'drivers/tspdrv' failed
make[1]: *** [drivers/tspdrv] Error 2
Makefile:950: recipe for target 'drivers' failed
make: *** [drivers] Error 2

stefant234, 2015-04-05:
Remove "-Werror-implicit-function-declaration" from the Makefile in your kernel source directory and try again

stefant234, 2015-04-05:
Kick -Wall out too while your on it. I never had much left for warnings which are treated as errors.. Unless you want to fix every little code derp :P

Xephi, 2015-04-06:
Tested without theses flags, but always these errors ( cause tspdrv seems to be always undeclared ^^")

stefant234, 2015-04-06:
Yeah, they gave us broken junk.. Theres a file /include/linux/dynamic_debug.h that is unusable.. I imported a clean one from the OnePlus One, which is also a SD801 phone to avoid any issues.. Compiles beyond those errors. Replace your file with this: https://github.com/stefant234/AndromedaCANCRO-KK/blob/master/include/linux/dynamic_debug.h

stefant234, 2015-04-06:
Don't know how I fixed tspdrv, it just stopped quirking :O

stefant234, 2015-04-06:
New place, same error.. Don't bother with my Import.. Something else is wrong here

arnabbiswasalsodeep, 2015-04-06:
there are a heck of errors in the source, fixed a lot of them and finally got it buildable but not bootable
try deleting this lines in the original immvibespi.c
>/* From Xiaomi start */
>                       pr_debug("drv2604 write 0x%02x, 0x%02x", data[i], data[i
>/* From Xiaomi end */

i m using the compiler arm-cortex_a15-linux-gnueabihf-linaro_4.9.1-2014.04 http://forum.xda-developers.com/showthread.php?p=59859361#post59859361

arnabbiswasalsodeep, 2015-04-06:
These are all the issues i faced and fixed and got it buildable
http://pastebin.com/F0fTegmq

i have got the zImage file but the problem is i dont know how to flash it and tried many methods but it didnt boot and went straight to fastboot

stefant234, 2015-04-06:
In the beginning of the compilation dtb's are generated. http://forum.xda-developers.com/showthread.php?t=2469510
You need to create the boot.img with the additional dt.img for it to boot

stefant234, 2015-04-06:
Already fixed most errors here apparently too.. Going to try your immvibespi.c fix now

arnabbiswasalsodeep, 2015-04-06:
i tried that but gave errors, anyways let it be! i got the z image to boot! YAY!!!

used this anykernel zip http://forum.xda-developers.com/showthread.php?t=2670512

stefant234, 2015-04-06:
Full success here too. Meh, I have a different way to assemble zips and boot.imgs :)

Xephi, 2015-04-06:
I will try theses changes :)  By the way,  is your kernel stable with CM-12 @stefant234 ?

stefant234, 2015-04-06:
I haven't tried, don't have a Mi4 anymore :/ Developing completely blind.. Would you like to give the build a shot? My usual test subjects are busy until the evening..

arnabbiswasalsodeep, 2015-04-06:
i m testing my current kernel on mokee cm12, working fine till now

DempaPower, 2015-04-06:
Is high vibration intensity fix with this one?

arnabbiswasalsodeep, 2015-04-06:
dont know about that one, can u explain
i used a different compression method for my kernel and its working fine except 2 issues
1- wifi is not turning on
2- slow paced animations have got laggy, the sppedy ones work fine

DempaPower, 2015-04-06:
Well,  for exemple in AOSP rom by Ivan for Mi3/Mi4, the vibration is really too much high. When you keep down volume key when you put in Silent mode, you can hear that your vibration does weird sound. It already killed some Mi3 vibrator of users...

arnabbiswasalsodeep, 2015-04-08:
hey, can any1 instruct me how to make the boot.img
the closest i have reached is there is an install.sh file in the arch/arm/boot which is used to make the kernel image but make install gives the following error
>make: *** No rule to make target `vmlinux', needed by `/Image'. Stop.

can any1 help me on how to get the script running?

AungThiha, 2015-04-12:
@arnabbiswasalsodeep video works fine. only having wifi issue.

arnabbiswasalsodeep, 2015-04-13:
then i hv to check cuz in my build the videos dont play, at first they start, then last for 1/4th of a sec and gives error cant play the video
i will do a facorty reset and try

emceethemouth, 2015-04-17:
I got everything booting with GCC 4.9.3, now all we need is the wifi drivers.

schspa, 2015-04-24:
Modify the Makefile of  drivers/tspdrv 
the line below will fix the build 
##EXTRA_CFLAGS += -I$(M)
EXTRA_CFLAGS += -I$(shell pwd)