MiCode/Xiaomi_Kernel_OpenSource #15: WLAN Prima Module Open Source Code for armani

Kra1o5, 2015-04-03:
As part of armani kernel source the wlan prima module code is missing.

Please release the it.

thewisenerd, 2015-04-06:
:+1: 

This is keeping us from using WiFi in the Stock ROM too.

alexret, 2015-04-17:
Hi.Some friends and I, have opened in mi.com to ask drivers http://en.miui.com/forum.php?mod=viewthread&tid=101360&extra=page=1&mobile=2
Also we are asking Hugo in Twitter to let go them
Please, help us

demonguy, 2015-05-26:
Hi, I think you could find a module in our phone, and adb pull it.
As far as i know, Xiaomi didn't open wifi driver source indeed, but you could retrieve it from a phone

liuguo09, 2015-05-27:
Please refer to this [FAQ](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/wiki/FAQ#where-is-wifi-driver-module)

thewisenerd, 2016-07-17:
Hi,

I wanted to ask you whether you have any  previous  experience  of this  type of work?  Here are some questions for you <http://cause.mckeegroupllc.com/e3yocb>

Speak to you later, contact.twn@openmailbox.org

thewisenerd, 2016-08-02:
Dear! 

We've had an amazing days out of town this  weekend, just take a look at the  place  we've been to, here  are some pictures  <http://control.phoenix-tech.org/e3qtcd>

Best wishes, contact.twn