MiCode/Xiaomi_Kernel_OpenSource #16: Miising WLAN Drivers Cancro

alexret, 2015-04-08:
I compiled the kernel and I have not wifi.
I've looked at drivers / Makefile and not the wifi

arnabbiswasalsodeep, 2015-04-08:
Same here too

KinG5Pac, 2015-04-09:
Can we just make module on automode in linaro ??

alexret, 2015-04-09:
I compile wiht Linaro 4.7, 4.8, 4.9, Sabermod 4.8, AK Linaro 4.9.3 and nothing, no wifi

AungThiha, 2015-04-12:
Same here, no wifi. I used linaro 4.9.3.

AungThiha, 2015-04-12:
Same here, no wifi. I used linaro 4.9.3. Have you guys overcome it?

arnabbiswasalsodeep, 2015-04-13:
i switched to bitti's repo
https://github.com/bittiscm12/android_kernel_xiaomi_cancro

AungThiha, 2015-04-13:
@arnabbiswasalsodeep how's everything with it. Is wi-fi alright?
As for me, I'm trying to add wlan driver from other device which have the same board. The wlan chipset is pronto for sure. I have tested HTC msm8974. It didn't work. Now, I'm going to test from https://github.com/tectas/android_kernel_lge_msm8974ac . still cloning. it's taking too long because of poor internet connection in Myanmar.

alexret, 2015-04-13:
I used driver wifi from Bitti, Audahadi and not work.
I renamed the wlan.ko that compiles to pronto_wlan.ko and does not work.
I have also taken prompt wlan.ko official and does not work
Sorry for my bad English

AungThiha, 2015-04-13:
Looks like Xiaomi have removed something. Something really hard to fork. How could they do this to us.

alexret, 2015-04-13:
They have been removed?  I do not see the changes

AungThiha, 2015-04-13:
No, what I meant is"they must have removed something before they released it to the public and keep the complete one away from being touched.

arnabbiswasalsodeep, 2015-04-13:
i m still trying to compile. i too have a slow connection of 1Mbps and on a pentium (goin to upgrade to i5 soon)

arnabbiswasalsodeep, 2015-04-13:
and no luck. btw are videos playing on the kernel? because on my device it doesnt work

AungThiha, 2015-04-13:
Yes, the video works fine. I used this toolchain.
https://www.mediafire.com/?dummbqmkaxai7

alexret, 2015-04-13:
Radio, mobile data, vídeos, games and everything in general works but stupid wifi no.
I've got everything in place and have made a flashable zip with my kernel 
For if it helps you
https://mega.co.nz/#!uFJRCb5R!RKCRRIv67fjmIQeAjX2I9uSFgfOc0ITGVcY1Oe6iuLI

arnabbiswasalsodeep, 2015-04-14:
i used the same toolchain. and the vids not playing is also a common bug for some mi3 users, same problem with rom built with source. downloading the flashable zip now. currently i am on ivans 5.1 3/April/15 as all the other roms built from kernel source give me the same video errors

arnabbiswasalsodeep, 2015-04-14:
@alexret From where u got those module files (the system folder)? and the method u used to make the boot image? the methods i use doesnt let the rom boot at all!

alexret, 2015-04-14:
I work on Miui 4.4.4, the modules are Miui.
If you do not believe me try it and see how start @arnabbiswasalsodeep

alexret, 2015-04-17:
Hi.Some friends and I, have opened in mi.com to ask drivers http://en.miui.com/forum.php?mod=viewthread&tid=101360&extra=page=1&mobile=2
Also we are asking Hugo in Twitter to let go them
Please, help us

AungThiha, 2015-04-17:
@alexret What's his e-mail address?

AungThiha, 2015-04-17:
Is this him?
https://twitter.com/hbarra

vasishath, 2015-04-17:
@alexret have you tried using my drivers ? Its here : http://github.com/vasishath/qcom_prima_xiaomi_cancro 
Its based on the same caf release tag as used by xiaomi. And i can confirm that ivan also uses this driver version ( i used strings to check the version Number from wlan.ko)
Refer to my post on xda here :
http://forum.xda-developers.com/xiaomi-mi-3/development/temasek-unofficial-cm12-t3012588/post59884938#post59884938

emceethemouth, 2015-04-17:
@vasishath Have you tried it? And does it work? Are you building inline or compiling the kernel by itself?

alexret, 2015-04-17:
Yes @AungThiha is Hugo.
I have not tried it, I just see.@vasishath Tomorrow if I have time I'll try it
I read if any try it

AungThiha, 2015-04-18:
@vasishath Which toolchain did you use? I used linaro 4.9.3 and I get this error.
```
drivers/staging/prima/CORE/HDD/src/wlan_hdd_main.c: In function ‘hdd_wlan_startup’:
drivers/staging/prima/CORE/HDD/src/wlan_hdd_main.c:8255:7: error: label ‘err_vos_nv_close’ used but not defined
       goto err_vos_nv_close;
       ^
```

vasishath, 2015-04-18:
@AungThiha @emceethemouth actually, i havnt compiled yet the driver or the kernel (i dont have the time right nw).. But i have seen audahadi use it in a recent rom and in that rom, people reported that wifi was working..

vasishath, 2015-04-18:
@AungThiha i have another release tag in my repo, it is a more recent release of the same version of the driver.. Maybe u could try that..

emceethemouth, 2015-04-18:
Here try this see if it's working: http://hubdroid.com/~Gnome/Stuxnet/Cancro/stuxnet-wlan4.zip

AungThiha, 2015-04-18:
Ok, I'll check them out.

romualdig, 2015-04-18:
Xiaomi, you will never fool us because we and many other people will never buy a Xiaomi! That's what you get treating this way your customers.

LuffyXDA, 2015-04-18:
why not you guys use wlan driver from CAF
https://www.codeaurora.org/cgit/quic/la/platform/vendor/qcom-opensource/wlan/prima/

alexret, 2015-04-18:
Same error as @AungThiha

alexret, 2015-04-18:
[image]
In Lollipop work fine.
I used this kernel https://github.com/kernel-hut/android_kernel_xiaomi_cancro

arnabbiswasalsodeep, 2015-04-19:
i am using the same source to build kernels. Will be using these fixes https://github.com/emceethemouth/kernel_cancro/commit/f5dcfd29b029deeafacb39b82bd5ff94cb437fee

demonguy, 2015-05-26:
Hi, I think you could find a module in our phone, and adb pull it.
As far as i know, Xiaomi didn't open wifi driver source indeed, but you could retrieve binary from official websites
And i've confirmed with my friends in Xiaomi, they're planning to put the driver binary to this repo soon

liuguo09, 2015-05-26:
Hi all, I've confirmed about this problem. And i'm sure the wifi driver is not developed by us.
I'm confused about it for long time and finally i found this driver is released by Qualcomm and they already opensource it.
I'll give you an instruction about how to get the codes after preparation.

liuguo09, 2015-05-27:
Please refer to this [FAQ](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/wiki/FAQ#where-is-wifi-driver-module)

emceethemouth, 2015-05-27:
Wow! @liugio09 that was helpful.

rooque, 2015-05-27:
@liuguo09 nice! Hope to see you more here

alone-wolf, 2015-06-06:
im more than new who can tell me how to deal with the wifi code

Vdragon, 2015-06-21:
Since the issue is solved, @alexret could you close this issue?

@MiCode

emceethemouth, 2015-06-23:
@Vdragon  before we close this, does anyone have successfully built the kernel and wlan source and had it worked on stock MIUI ROM? Coz as far as I know, all custom kernel devs are building for CM Based ROMs, and the kernel should be packed with the required modules that are on the ROM, in a nut shell, you cant flash the kernel by itself without the modules. Now, if anyone can confirm that by building the source code with the wlan source is working fine out of the box without the required modules. Then please, by all means tell us. 

Thanks,

Jean Francis Dominique S. Mabalot <jmabalot@google.com>

Vdragon, 2015-06-23:
@emceethemouth I didn't have the device thus I can't test it, I'm fine with verifying the solution though.

Ｖ字龍 &lt;Vdragon.Taiwan@gmail.com&gt;

emceethemouth, 2015-06-23:
Just read the FAQ and it seems downloading the whole source is  a pain that's why some people won't try it. Well here's a much easier thing to do:

- git clone git://codeaurora.org/platform/vendor/qcom-opensource/wlan/prima prima
- cd prima
- git checkout LNX.LA.3.5.2.2.1-04310-8x74.0
- cd ../
- then cp prima ~/$CANCRO_SOURCE_CODE/drivers/staging

Then build the kernel, much easier that way.

shellbombs, 2017-04-19:
i've flashed the boot.img of CM13(android 6.0) with MIUI7(android 6.0), wifi not available.
i've also compiled this kernel, and replaced it with the boot.img of MIUI7(android 6.0) using mktools, boot failed.