MiCode/Xiaomi_Kernel_OpenSource #18: Please comply to terms of GNU GPL2 License, release Mi Pad's Linux© kernel source code! | 请遵从 GNU GPL2 授权条款的规定，释出小米平板的 Linux© 作业系统内核来源代码！ | 請遵從 GNU GPL2 授權條款的規定，釋出小米平板的 Linux© 作業系統核心來源程式碼！

Vdragon, 2015-04-26:
Linux© kernel is a copyrighted software written by @torvalds et al. and is released using GNU GPLv2 license.  According to licensing terms authors of derived work of Linux kernel must provide modified Linux© source code along with the distribution of the software.  Since Mi Pad uses Linux kernel and is distributed to the public @MiCode must comply to terms of GNU GPL2 License and release Mi Pad's Linux© kernel source code.

Vdragon, 2015-04-26:
Hi everyone concerning with this issue, please contribute other languages title(& issue summary also) so we can get more attention from @MiCode ;)

omarhasan95, 2015-05-07:
Provide us with mipad kernel sources

vartom, 2015-05-07:
Agreed. Provide us with mipad kernel sources.

mikoda, 2015-05-07:
totally agree

TheShinobi, 2015-05-07:
Yes, release the kernel sources for the Xiaomi Mipad!

omarhasan95, 2015-05-08:
Why there is no reply from mi???

abhiyoyo, 2015-05-08:
Yes they must provide kernel sources of Mipad keeping in mind of customer interest if they can make only half baked miui ROM then give sources many talanted devs will make stable Roms . Keep your image clean your biggest con is closed source . that why you were denied place in Xda forum . Its good to see many Mi mobiles got sources in recent times . But why not Mipad . if its something nvidia does not want you to give or what ?? Please just be cleat why its not possible or when its possible . Your silence on this issue is hurting more than unavailability of sources . we are still not sure whether we are banging our head needlessly or at wrong place

Doraha-Raman, 2015-05-08:
yes , they should have already released kernels

KinG5Pac, 2015-05-09:
I don't think they will care about this. Same scenario was on mi3. And @Hugo again lied about kernel release , I'm about 3 month kernel release fro all devices.

abhiyoyo, 2015-05-09:
but @KinG5Pac  but they released sources for mi3 eventually

KinG5Pac, 2015-05-09:
@abhiyoyo 2-3 years old phone. And u say it released ?

abhiyoyo, 2015-05-09:
Yes

co11ider, 2015-05-09:
Yes. Dont make me regret buying a mipad and will never ever buy a Xiaomi product again. Also I'm suspecting you guys must have hidden some dirty secret inside it, like the Indian goverment warned.

JuanMiguelML, 2015-05-10:
We need kernel source! Provide us the MiPad kernel source!

sharifsonic, 2015-05-15:
YEAH GUYS, Dont defame your self.
I used to be a great MI fan..
I'm losing it now!!
Release the mipad kernel.

liuguo09, 2015-05-28:
We will, but i can't give you an accurate time

Vdragon, 2015-05-28:
@liuguo09 What about your *will* on issue #4 ?

abhiyoyo, 2015-05-28:
ok sir thanks for reply @liuguo09  sir atleast an eta ?? will we get within 1 month ??

stefant234, 2015-05-28:
He said that he can't give you a time.. If you work in software development you should understand how useless ETAs are

abhiyoyo, 2015-05-28:
@stefant234 chill he said exact time not eta Sir !!!!!!

omarhasan95, 2015-05-28:
Only sources matter to us please provide us asap we dont need eta

abhiyoyo, 2015-05-28:
@omarhasan95 hope so but I know it takes lil time but don't know 1 month should be enough again I don't work in software development so no exact idea but redmi 1s and note 4g can get quick so hope Mipad will be quick

stefant234, 2015-05-28:
I am relaxed. Just out of curiosity, who of you do kernel development? Bombarding the poor man won't help anyone and won't change company policy. I'm also not saying that you force anyone, I just tell you that any ETA in software development is not to be trusted, because in 99.9% of the time, they can't be held. Thats what I meant. Since thus is public I can jump in as I please ;)

abhiyoyo, 2015-05-28:
OK jump hehe but its not bombarding lol its user requests @stefant234 and poor man ?? Lol we just asking not throwing grenades haha

stefant234, 2015-05-28:
Imagine being in his position. Xiaomi sent him to maintain their Github and now a few doyen people see him as the personification of GPL violations :D Anyways, this sourcecode's state is why I abandoned Xiaomi and going to stick with Samsung for a while. At least they provide the sourcecode for every firmware update :)

abhiyoyo, 2015-05-28:
Man @stefant234 people are not targeting him they target xiaomi only and yes its GPL violation and this is the page we really can ask about sources lol who even said he violated this post was about xiaomi :D

stefant234, 2015-05-28:
I think our problem is English, you apparently don't fully get me while I don't fully get you either :D

abhiyoyo, 2015-05-28:
Ok hehe

COI3RA, 2015-06-03:
Please release Mipad's Kernel Source code

museroom, 2015-06-10:
A China style riot is required for a China company to listen to their customers.

omarhasan95, 2015-06-11:
Release the sources asap its more than one year...... Before lauching new devices everymonth please take care of your pre released devices

villarhdez, 2015-06-12:
Por favor, no hagan que me arrepienta de haber comprado mi primer dispositivo Xiaomi y liveren el kernel. No les cuesta nada, solo haganlo o al menos desarrollen una ROM lo suficientemente estable para la Mipad

Vinnipeg, 2015-06-16:
I join the request, +1

nightfury858, 2015-06-17:
I'm really disappointed. I don't think I'm gonna buy another Xiaomi device after I'm done with my Mi Pad. At least not until the kernel sources are released. So please release Mipad's Kernel Source code soon.

TrickNL, 2015-06-17:
Xiaomi please release the kernel source files so we al can get creative and start creating roms to get the most out of our Mi pad. Thank you in advance 😁

Motivate you're customers so in the future they buy more Xiaomi devices. Surely you want this and not the other way around. The device is a year old now and we have been suffering with your unoptimized Mi ui rom.

akb1, 2015-06-18:
Thats how they do it. Welcome to China brothers. If it is in their favour they will release it or you forget about it. They don't give a damn about copyright or anybody's rights. But then again, if they don't they underestimate its consequences for their company.  Release the Kernel.

LordHexd, 2015-06-26:
This may be mainly for the MiPad but as a new Mi4i user who bought into the hype of a great device at great price, I too am eagerly waiting for the kernel release for the device. I just can not use MIUI for the bloat and only 16GB of storage available. Fix this and I may have reasons to buy more of Mi/Yu products. As of now I'm very happy with the MiBand and bought 7 for my friends and family as I wanted them to feel the joy of owning a product as well built and functional as it is. MiPowerBanks are just as good and the quality and function meets the standard I expect from the product. Unfortunately my latest purchase has me second guessing whether the Mi team thought through the full scope when designing the phone itself and the ever so bloated MIUI.

doomboss, 2015-06-27:
I know the developer from Xiaomi are very busy, but please just release the kernel source! I really hope there will be more development and the Mipad definitely has more potential if you guys can let outside developers to work on their custom ROM with the kernel source!

jockiller, 2015-07-02:
快来放源码吧…不开放你就给我们正了八经的开发MIUI

skr4tchGr3azyMonkiBallllllZzzz, 2015-07-06:
Let get together and sue Xiaomi for violating GPL.

skr4tchGr3azyMonkiBallllllZzzz, 2015-07-06:
MIUI on MIPAD SUCKS BALLz!!!

skr4tchGr3azyMonkiBallllllZzzz, 2015-07-06:
Boycott MiPad NOW!

kirill0v, 2015-07-23:
Give us at least some decent reply with status or a legitimate reason for not releasing it! 

Guys sign this if you haven't already, a petition asking to release kernel source: https://goo.gl/6xbZoY

rooque, 2015-07-23:
Support the cause here http://en.miui.com/forum.php?mod=viewthread&tid=136345

d3m3vilurr, 2015-08-05:
@liuguo09 can you share any news?

moonges, 2015-08-18:
Guys please! we need Mi Pad's Linux© kernel source code!

raftaarsingh, 2015-08-20:
plz provide us with kernel source code of mi pad its enough time than expected

abhiyoyo, 2015-08-21:
@liuguo09  please man its enough time please release sources for betterment of mipad itself

JuanMiguelML, 2015-09-04:
Xiaomi, release the kernel sourde now! We need it!

jza70, 2015-09-07:
Release the sources and get rid of spyware. You are violating the license.

Yrahate7, 2015-09-10:
Enough of this s###. Xiaomi isn't even updating the device to miui7 and neither are they releasing source codes... 
If this continues then I will shift over to the developer friendly nexus 9.
Expensive  yet worth it. 
Better release the codes for sake of 50000 MI pad users

sandipCS, 2015-09-19:
release the kernel source please...we need it.

Ash-Drake, 2015-09-29:
please release the kernel sources xiaomi, we need it, we only get mostly unstable rom from mi, we users deserve better experience, and you know it, please don't make it harder for us to survive in this device. please release the sources soon for you to make it up for giving us such an incredible hardware

Vdragon, 2015-09-29:
### UPDATE
* Remove the politeness part from the title as this is not a *friendly* request.
* Add some tags that make the issue more **serious** and hopefully, alert someone that works in XiaoMi's LEGAL department(If there's really one, of course)

skr4tchGr3azyMonkiBallllllZzzz, 2015-10-03:
I'm willing to fund a campaign to sue Xiaomi in China. What legal actions can be taken?

abhiyoyo, 2015-10-03:
Thanks for effort @skr4tchGr3azyMonkiBallllllZzzz  they will learn a lesson

Vdragon, 2015-10-04:
@skr4tchGr3azyMonkiBallllllZzzz You'll need to have someone holding Linux kernel's copyright to sue XiaoMi, this means someone that previously committed their code into Linux kernel before XiaoMi uses it as MiPad's kernel.

tranphungan, 2015-10-06:
Still waiting for Kernel release, thank for releasing asap

d3m3vilurr, 2015-10-10:
ok, few hours ago, armani-kk kernel pushed. (#10) but mipad and other devices's kernel still missing. boo....
@liuguo09 r u there?

abhiyoyo, 2015-10-10:
Yes Mipad should also get it it has been long wait   @liuguo09  you told long ago sources for Mipad will be released I think time has come make Mipad users happy as you did for armani

[77 more replies]

closed 2018-02-26 by Vdragon