MiCode/Xiaomi_Kernel_OpenSource #22: Please Release Xiaomi Redmi 2 Kernel source code

St1cker, 2015-06-03:
GPL

Vinnipeg, 2015-06-16:
Join the request, +1

pirej, 2015-07-14:
we are waiting..

rooque, 2015-07-14:
Don't know why they delay it , what they have to lose ?

thePanz, 2015-07-17:
Please take a look at: https://github.com/zwliew/android_device_xiaomi_hm2014811/issues/30
The repository is for 2014811, but the developer says it's Ok also for 2014813 and other variants.

rooque, 2015-07-17:
@thePanz device source != kernel source

thePanz, 2015-07-17:
@rooque: sorry, my misunderstanding.
@zwliew already clarified it: https://github.com/zwliew/android_device_xiaomi_hm2014811/issues/30#issuecomment-122295055

balika011, 2015-07-17:
We need only the sound folder form the source. All other driver is done and working by me. :smile:

emfox, 2015-07-20:
+1
@balika011 so can you build lollipop for redmi 2 now, perfectly run everything just without sound?

balika011, 2015-07-21:
@emfox 64bit mode boot not works at all. I dunno why, but we can't boot 64bit code. We have an "msm8916 qrd skum" soc. No public info about that on the net. The code for "msm8916 qrd skui" soc boots up well in 32bit mode, but not in 64. :disappointed: 
The back speaker works fine. The headset works whit a hack. The font speaker not works at all.I need more time to find the changes, but i don't have any as of now. Maybe some weeks later.

rooque, 2015-07-23:
Support the cause here http://en.miui.com/forum.php?mod=viewthread&tid=136345

kumajaya, 2015-08-06:
@balika011 I managed to reconstruct "Qualcomm Technologies, Inc. MSM 8916 QRD SKUM" dts from Redmi 2 device tree blob. Is it useful for you?

balika011, 2015-08-06:
i already did it weeks ago. i have a fully working kernelsource, wating to be pushed.

kumajaya, 2015-08-07:
Great! Impatiently waiting for your source.

yipperr, 2015-08-08:
Anyone knows if the wireless driver supports monitor mode ? Also anyone has info in the wlan chipset driver used

balika011, 2015-08-11:
https://github.com/miuihu/android_kernel_xiaomi_redmi2

kumajaya, 2015-08-23:
@balika011 I merged your patch to upstream Cyanogen MSM8916 kernel source https://github.com/kumajaya/android_kernel_cyanogen_msm8916/tree/cm-12.1-wt88047 Still in progress but I confirm using it as source, TWRP and Cyanogen recovery boot Redmi 2 successfully. A small changes in dtsi fix LCD garbled issue I experienced before.

nick37332001, 2015-10-12:
https://www.change.org/p/mr-lei-jun-mr-hugo-barra-mr-dexter-ren-xiaomi-inc-xiaomi-indonesia-release-kernel-source-code-for-redmi-2

balika011, 2015-10-15:
We already have it :smile:

ZuBB, 2015-10-15:
@balika011 what do we have?

balika011, 2015-10-15:
Source?

ZuBB, 2015-10-15:
so 5.1/6.0 would not be a problem for our device?

balika011, 2015-10-15:
5.1 works fine afaik

emfox, 2015-10-15:
5.1 not so perfect as I know, some compass chip (yas037 or what?) and some battary controller driver not available, and there will be some random reboot with heavy usage.

ZuBB, 2015-10-15:
so we do not  have *all* the stuff

balika011, 2015-10-15:
bc is not used at all in stock kernel(no driver), yahama compass is the same, missing.

jawircodes, 2015-10-30:
Please share tutor. step by step to build this source .

nick37332001, 2015-10-30:
@balika011 We spent so much time trying to debug everything and patch it up. If only we had the full kernel source in the first place, we would be able to work on features instead of spending time wrecking our minds to fix a problem.

mits619, 2015-11-16:
Release the Kernel source for Redmi 2 Prime

jawircodes, 2015-11-21:
until now has not released the source code!!!! HOW its..

future-geek, 2016-01-17:
+1
@balika011 This https://github.com/YUPlayGod/android_kernel_yu_msm8916 might help you in reproducing msm8916 qrd skum for 64 bit

ghwijaya, 2016-01-19:
+1 Redmi 2 and Redmi 2 Prime

Adipatiarya, 2016-01-31:
where is the kernel source????

Xily9, 2016-02-11:
+1.Please open the kernel source code.