MiCode/Xiaomi_Kernel_OpenSource #25: Dior missing camera drivers

haoyangw, 2015-08-04:
There is obviously missing camera drivers for the back camera(Sunny Q13s01B), after I built the kernel from source and use it to boot CM12 only front camera works. I searched through the kernel source and found only one declaration of 'q13s01b' in the dts files(which do not in itself make up a camera driver -.-) while there are 3 declararions of 'p5v40a'(front camera) including c code to detect and perform operations on the front camera. This is probably why front camera works but not back(I've already included ALL proprietary camera libs). Plase DO fix this, armani doesn't have this issue! It's really annoying to only have front camera working, and almost impossible for a single dev not affiloated with xiaomi to fix it himself!

kirainmoe, 2015-08-10:
Last commit of this Project is on 30 Dec 2014 so it semms that nobody is watching this for a long time :(
And, I just a common user of Dior, I am sorry that I can't help you XD

kumajaya, 2015-08-20:
You can extract the dtb file and convert it to dts as a reference.

kumajaya, 2015-10-11:
Both front and back camera working on Redmi 2 Lollipop port without a crap of official kernel source! #ShameOnYouXiaomi

haoyangw, 2015-10-16:
And one of the devs on xda fixed the back camera! :) his name is audahadi, check out his github haha #xiaomisucks

haoyangw, 2015-10-16:
#xiaomiuseless

kumajaya, 2016-01-01:
Marshmallow port for Redmi 2 without any major bug, without vendor kernel source. Stop monetize Linux! #shameonyouxiaomi #xiaomiuseless

nick37332001, 2016-01-01:
@kumajaya no use... they dont reply