MiCode/Xiaomi_Kernel_OpenSource #27: Please comply to terms of GNU GPLv2 License, release Redmi (Note 3g)'s Linux© kernel source code!／请遵从 GNU GPLv2 授权条款的规定，释出红米手机 (Note 3g) 的 Linux© 作业系统内核来源代码！／請遵從 GNU GPLv2 授權條款的規定，釋出紅米手機 (Note 3g) 的 Linux© 作業系統核心來源程式碼！ #解放小米 #LibreXiaoMi #4

git-rbanerjee, 2015-08-21:
Dear @MiCode, hehao hehao@xiaomi.com, Hugo Barra and XiaoMi corp.,

English version

Linux© kernel is a copyrighted software written by @torvalds et al. and is released under the GNU GPLv2 license. According to licensing terms distributor of derived work of Linux kernel must provide modified Linux© source code along with the distribution of the software. Since Redmi (Note 3g) uses Linux kernel and is distributed to the public @MiCode must comply to terms of GNU GPL2 License and release Redmi (Note 3g)'s Linux© kernel source code.

简体中文（中国大陆地区）版本
Chinese(Simplified)(China) version

Linux© 作业系统内核是由 @torvalds 等人设计的一个享有智慧财产权的软件并其被已 GNU GPL 第 2 版授权条款释出。根据授权条款规定基于 Linux© 作业系统内核的衍伸作品的散布者必需伴随其散布的软件提供被修改的 Linux© 作业系统核心来源代码。既然红米手机使用了 Linux© 作业系统内核且已被散布至公开场合 @MiCode 必须尊从 GNU GPL 第 2 版授权条款的规定释出红米手机的 Linux© 作业系统核心来源代码。

正體中文（台灣地區）版本
Chinese(Traditional)(Taiwan) version

Linux© 作業系統核心是由 @torvalds 等人設計的一個享有智慧財產權的軟體並其被已 GNU GPL 第 2 版授權條款釋出。根據授權條款規定基於 Linux© 作業系統核心的衍伸作品的散佈者必需伴隨其散佈的軟體提供被修改的 Linux© 作業系統核心來源程式碼。既然紅米手機 (Note 3g)使用了 Linux© 作業系統核心且已被散佈至公開場合 @MiCode 必須尊從 GNU GPL 第 2 版授權條款的規定釋出紅米手機 (Note 3g) 的 Linux© 作業系統核心來源程式碼。

Seriously, your Redmi Note 3g customer.

nick37332001, 2015-09-24:
Agreed... Xiaomi please release all kernel sources as well as MediaTek ones as you are violating these rules.

git-rbanerjee, 2015-10-11:
Xiaomi where is the kernel sources, its been an year now you released the device.

git-rbanerjee, 2015-10-11:
@liuguo09 can you please release Redmi Note 3g ,lcsh kernel sources ???

nick37332001, 2015-10-12:
https://www.change.org/p/mr-lei-jun-mr-hugo-barra-mr-dexter-ren-xiaomi-inc-xiaomi-indonesia-release-kernel-source-code-for-redmi-2

amolpingle, 2015-10-12:
Please release Redmi Note 3g, lcsh kernel sources, please.

nick37332001, 2015-10-13:
http://www.xda-developers.com/gplv2-and-its-infringement-by-xiaomi/

git-rbanerjee, 2015-10-25:
Still waiting for the kernel sources.

git-rbanerjee, 2015-10-25:
@liuguo09

wonson, 2015-10-29:
It has been about a year after you promised last October to release kernel in Feb 2015.

git-rbanerjee, 2015-11-12:
Still waiting for source code.

nick37332001, 2015-11-12:
screw you xiaomi!!! We have fully working marshmallow on redmi 2 without your kernel.
Just give up and let us have the sources!!!

git-rbanerjee, 2015-11-13:
@liuguo09 ,can we have any words ??

ICHx, 2016-03-25:
its mtk, just forget about it. (or you can give mtk a "what" )