MiCode/Xiaomi_Kernel_OpenSource #33: Please, release kernel source from Redmi Note 2

XGenDev, 2015-12-13:
Hello, this will be great forma developers. Do it for the comunity, because we love xiaomi and que trust you.

XGenDev, 2015-12-13:
More than 500 buddies sign here: https://www.change.org/p/xiaomi-inc-mi-com-hugo-barra-release-kernel-source-from-xiaomi-redmi-note-2?recruiter=445660762&utm_source=share_petition&utm_medium=copylink

GaMeRaM, 2016-02-07:
More than 1000 buddies signed here: https://www.change.org/p/xiaomi-inc-mi-com-hugo-barra-release-kernel-source-from-xiaomi-redmi-note-2

GaMeRaM, 2016-02-24:
More than 1200 buddies signed

sm4rk0, 2016-03-06:
1500 and counting...

sarvesvara, 2016-03-07:
Who is checking the counts, really?

veaceslav, 2016-06-07:
Please release the kernel sources.

everdeath666, 2016-08-24:
Why not realese some files?!?!? Kernel source is almoust done in this thread ! https://github.com/nofearnohappy/Custom_kernel_hermes_new

Give some need files for stabble work!