MiCode/Xiaomi_Kernel_OpenSource #36: Thomas Error Compiling

ocoot, 2016-03-03:
hi guys, i faced new error here for thomas. maybe anyone can help me?

DTC     arch/arm/boot/dts/qcom/../msm8226-sim.dtb
Error: arch/arm/boot/dts/qcom/msm8226-regulator.dtsi:532.2-3 label or path, 'pm8226_chg_chgr', not found
FATAL ERROR: Syntax error parsing input tree
/android/miui/arch/arm/boot/dts/qcom/Makefile:173: recipe for target 'arch/arm/boot/dts/qcom/../msm8226-sim.dtb' failed
make[1]: *** [arch/arm/boot/dts/qcom/../msm8226-sim.dtb] Error 1
/android/miui/arch/arm/Makefile:307: recipe for target 'dtbs' failed
make: *** [dtbs] Error 2
make: *** Waiting for unfinished jobs....

deeplyarnab, 2016-03-03:
`msm8226-regulator.dtsi:532.2-3 label or path, 'pm8226_chg_chgr', not found`

Check if that file is present, otherwise you downloaded incomplete source

ocoot, 2016-03-03:
its missing. and i added it manually one hour ago :3 thanks for reply btw and now waiting the other errors

ocoot, 2016-03-03:
how about this?

drivers/soc/qcom/subsystem_restart.c: In function 'subsys_register':
drivers/soc/qcom/subsystem_restart.c:1343:19: error: 'WT_SUBSYSTEM_REASTART_LEVEL' undeclared (first use in this function)
  if (0 == strncmp(WT_SUBSYSTEM_REASTART_LEVEL, "system", 6)) {
                   ^
drivers/soc/qcom/subsystem_restart.c:1343:19: note: each undeclared identifier is reported only once for each function it appears in
scripts/Makefile.build:308: recipe for target 'drivers/soc/qcom/subsystem_restart.o' failed
make[3]: *** [drivers/soc/qcom/subsystem_restart.o] Error 1
scripts/Makefile.build:455: recipe for target 'drivers/soc/qcom' failed
make[2]: *** [drivers/soc/qcom] Error 2
scripts/Makefile.build:455: recipe for target 'drivers/soc' failed
make[1]: *** [drivers/soc] Error 2
make[1]: *** Waiting for unfinished jobs....

jawircodes, 2016-03-03:
i am success build this source... check me build...
and try.
http://46.166.164.206/~jawircodes/

deeplyarnab, 2016-03-03:
U have to edit makefile to ignore some basic warnings.
I am assuming you are trying to build the kernel alone

Refer this
https://github.com/kernel-hut/android_kernel_xiaomi_cancro/commit/aff2f67f47c264cd1ff43f68573b58951a76dbaa

ramsudharsan, 2016-03-04:
I managed to work on the errors present in the kernel source and have made it to compile. I can link you to my alpha source code :rabbit: I have used the original defconfig (msm8916_defconfig) for my build which seems pretty bloated.  As I don't have a device, I can't do much work for this device. I hope my kernel source atleast acts a base reference for starting.

Link to the source: https://github.com/ramsudharsan/android_kernel_xiaomi_thomas

P.S. I have used arm-eabi-4.7 cross compiler from Google to compile the kernel.
P.P.S. @deeplyarnab I don't think removing gcc wrapper is ideal. Do you think all the manufacturers who tweak the kernel source for their own products might not have noticed that? Its simply like we are moving away from the errors/warnings instead of facing it. I myself have used those commits in the past but now, imo, it is not ideal. Do correct me if I am wrong.

deeplyarnab, 2016-03-04:
You are completely right @ramsudharsan
I provided that fix cuz i am guessing he is new to yhe development and hence won't be completely be able to fix it as it requires tedious work, that was just a quick fix to errors so he could try n learn more. Thanks for pointing it out to people.

jawircodes, 2016-03-04:
using msm8916-perf_defconfig edit look this is
/*if (0 == strncmp(WT_SUBSYSTEM_REASTART_LEVEL, "system", 6)) {
		subsys->restart_level = RESET_SOC;
	}

	if (0 == strncmp(WT_SUBSYSTEM_REASTART_LEVEL, "related", 7)) {
		subsys->restart_level = RESET_SUBSYS_COUPLED;
	}*/

ramsudharsan, 2016-03-05:
@jawircodes It should not be commented. Instead I have implemented in another way. See here: https://github.com/ramsudharsan/android_kernel_xiaomi_thomas/commit/ddc4d2c3dd5a551ef95b7dda3198dbcf4ae0301b#diff-8e3ed10d54de9baa5c023449807590c8R45

jawircodes, 2016-03-05:
@ramasudharsan. thanks.. i will try your method...

ocoot, 2016-03-05:
@ramsudharsan i faced an brightness and wifi bug -_-

ramsudharsan, 2016-03-05:
@ocoot The wifi bug is because of lack of prima drivers in kernel. I will work on adding it tonight. I think the Xiaomi uses the same kernel tag for its prima modem. Else we have to do some trial and error method. And for brightness I have to review the defconfig. I will do these tonight and push it.

ocoot, 2016-03-05:
@ramsudharsan thankyou, waiting for it :) :+1:

jawircodes, 2016-03-05:
wifi not work...please fix it..

ramsudharsan, 2016-03-05:
@jawircodes @ocoot I have tried to resolve all warnings and errors that showed up after merging prima drivers. Don't forget to copy the *.ko files after compiling. If you guys don't like it, I will put everything inbuilt. Check my repo for the updated kernel tree.

xnickbrandtx, 2016-03-06:
@ramsudharsan did you get this?
ERROR: modpost: Found 2 section mismatch(es).
To see full details build your kernel with:
'make CONFIG_DEBUG_SECTION_MISMATCH=y'
To build the kernel despite the mismatches, build with:
'make CONFIG_NO_ERROR_ON_MISMATCH=y'
(NOTE: This is not recommended)
/home/nick/android/thomaskernel/scripts/Makefile.modpost:95: recipe for target 'vmlinux.o' failed
make[1]: *** [vmlinux.o] Error 1
Makefile:788: recipe for target 'vmlinux' failed
make: *** [vmlinux] Error 2

ramsudharsan, 2016-03-06:
@xnickbrandtx Nope I didn't get any such error. Is it from my kernel source that you are getting this error?

xnickbrandtx, 2016-03-06:
@ramsudharsan No... I regenerated the defconfig from msm8916-perf_defconfig

ramsudharsan, 2016-03-06:
Do this make CONFIG_DEBUG_SECTION_MISMATCH=y as your compiler asks you and you will know what exactly is troubling inside. :hamster:

xnickbrandtx, 2016-03-06:
@ramsudharsan I'll just try your defconfig see if this error will occur again or not

xnickbrandtx, 2016-03-06:
@ramsudharsan any ideas?
WARNING: drivers/built-in.o(.data+0xd0cc): Section mismatch in reference from the variable smd_xprt_cfg to the function .init.text:mpm_irq_domain_linear_size()
The variable smd_xprt_cfg references
the function __init mpm_irq_domain_linear_size()
If the reference is valid then annotate the
variable with __init* or __refdata (see linux/init.h) or name the variable:
*_template, *_timer, *_sht, *_ops, *_probe, *_probe_one, *_console

WARNING: drivers/built-in.o(.data+0xd0e4): Section mismatch in reference from the variable smd_xprt_cfg to the function .init.text:mpm_irq_domain_legacy_size()
The variable smd_xprt_cfg references
the function __init mpm_irq_domain_legacy_size()
If the reference is valid then annotate the
variable with __init* or __refdata (see linux/init.h) or name the variable:
*_template, *_timer, *_sht, *_ops, *_probe, *_probe_one, *_console

  LD      net/wireless/cfg80211.o
  LD      net/wireless/built-in.o
  LD      net/built-in.o
  LINK    vmlinux
  LD      vmlinux.o
  MODPOST vmlinux.o
WARNING: vmlinux.o(.data+0xed8c): Section mismatch in reference from the variable smd_xprt_cfg to the function .init.text:mpm_irq_domain_linear_size()
The variable smd_xprt_cfg references
the function __init mpm_irq_domain_linear_size()
If the reference is valid then annotate the
variable with __init* or __refdata (see linux/init.h) or name the variable:
*_template, *_timer, *_sht, *_ops, *_probe, *_probe_one, *_console

WARNING: vmlinux.o(.data+0xeda4): Section mismatch in reference from the variable smd_xprt_cfg to the function .init.text:mpm_irq_domain_legacy_size()
The variable smd_xprt_cfg references
the function __init mpm_irq_domain_legacy_size()
If the reference is valid then annotate the
variable with __init* or __refdata (see linux/init.h) or name the variable:
*_template, *_timer, *_sht, *_ops, *_probe, *_probe_one, *_console

ocoot, 2016-03-06:
@ramsudharsan finished compiling. waiting someone to test now

xnickbrandtx, 2016-03-06:
@ramsudharsan I just realized i made a huge mistake. I used the wrong compiler :p

xnickbrandtx, 2016-03-06:
@ramsudharsan do you want to join in github.com/thomas-dev ? I currently trying to bring over caf LA.BR.1.2.6_rb1.8

ramsudharsan, 2016-03-06:
@xnickbrandtx I have already joined the organization. I am bringing up kernel from LA.BR.1.2.7_rb1.5

xnickbrandtx, 2016-03-06:
@ramsudharsan hmm 1.2.6 should be better since its used in cyanogenmod's msm8916 repo

ramsudharsan, 2016-03-06:
@xnickbrandtx Though either family will probably work.The 1.2.7 family is a bit newer than 1.2.6 (though both are maintained).

xnickbrandtx, 2016-03-06:
@ramsydharsan do you have telegram? We can talk there instead

ramsudharsan, 2016-03-06:
Nope. I don't use my mobile for chatting. Hangouts are preferred. Email: ramsudharsanm@gmail.com

jawircodes, 2016-03-06:
@ramsudharsan, i follow your instruction , but wifi and brightness still is not working..

jawircodes, 2016-03-06:
zImage too big "7.5mb" .in stock rom zImage "6.4mb".

ramsudharsan, 2016-03-06:
@jawircodes Can you get me logs?

ocoot, 2016-03-11:
like this? @ramsudharsan this logs its not from me. bcz i'm not using miui and testing my self the kernel. http://pastebin.com/i8frcpvV
sorry for late reply

ramsudharsan, 2016-03-11:
@ocoot Kernel specific logs are appreciated. Try pulling /proc/kmsg file.

dev-elixir, 2016-04-14:
@ramsudharsan Compiled the kernel and booted it but a black screen appeared after the boot and also am unable to take logs here is the error on adb logcat "Unable to open log device '/dev/log/main': No such file or directory"  .. I checked if it is on or notby calling on the phone and the call rings but nothing is showing..

dev-elixir, 2016-04-14:
@ramsudharsan after sometimes it got rebooted... any idea

dev-elixir, 2016-04-14:
I hanged the call still it was ringing

dev-elixir, 2016-04-15:
Attached kmsg
[kmsg.tx.zip](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/files/220155/kmsg.tx.zip)

dev-elixir, 2016-04-15:
[kmsg.txt](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/files/220157/kmsg.txt)

jawircodes, 2016-05-11:
bad source. the Result of camera is not the same as the default MIUI stock    .

dev-elixir, 2016-05-11:
@jawircodes 
Kernel got booted ???

jawircodes, 2016-05-11:
yes..

dev-elixir, 2016-05-11:
@jawircodes need help do u use any messaging service ???

xnickbrandtx, 2016-05-11:
Well the kernel itself is an older version compared to the one build by miui.

jawircodes, 2016-05-11:
@dev-elixir
share your branch..

dev-elixir, 2016-05-11:
@jawircodes https://github.com/dev-elixir/Xiaomi_Kernel_OpenSource/tree/thomas-kk-oss

jawircodes, 2016-05-11:
@dev-elixir
this is for miui rom or cyanogen.

dev-elixir, 2016-05-12:
@jawircodes miui ofcourse

eebssk1, 2016-06-10:
drivers/soc/qcom/subsystem_restart.c: In function 'subsys_register':
drivers/soc/qcom/subsystem_restart.c:1480:19: error: 'WT_SUBSYSTEM_REASTART_LEVEL' undeclared (first use in this function)
  if (0 == strncmp(WT_SUBSYSTEM_REASTART_LEVEL, "SYSTEM", 6)) {
                   ^
drivers/soc/qcom/subsystem_restart.c:1480:19: note: each undeclared identifier is reported only once for each function it appears in
scripts/Makefile.build:308: recipe for target 'drivers/soc/qcom/subsystem_restart.o' failed

anyway to solve it?