MiCode/Xiaomi_Kernel_OpenSource #38: Source request for gemini (Xiaomi Mi5) Kernel

h2o64, 2016-04-02:
Hello,
the Xiaomi mi5 device is out since a month (http://en.miui.com/thread-230807-1-1.html)
This build is available from xiaomi servers themselves (http://en.miui.com/download-299.html) :
```
ro.build.id=MRA58K
ro.build.display.id=MRA58K
ro.build.version.incremental=V7.2.8.0.MAACNDB
ro.build.date=Tue Mar 22 18:36:59 CST 2016
ro.product.model=MI 5
ro.product.brand=Xiaomi
ro.build.product=gemini
ro.build.description=gemini-user 6.0 MRA58K V7.2.8.0.MAACNDB release-keys
ro.build.fingerprint=Xiaomi/gemini/gemini:6.0/MRA58K/V7.2.8.0.MAACNDB:user/release-keys
```

According to GPL 3.0 agreements (http://www.gnu.org/licenses/gpl-3.0.en.html), could Xiaomi release the kernel sources quickly ?

anarkia1976, 2016-05-01:
This is true, we need source! please.

teknoraver, 2016-05-15:
sources, please

a mi 5 owner

scafroglia93, 2016-05-29:
Source Source Source Source

anarkia1976, 2016-05-29:
Source is necessary, otherwise you loose xiami! thanks.

arter97, 2016-05-29:
Source Source Source Source(2)

teknoraver, 2016-05-29:
`**********************************************************
**   NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE   **
**                                                      **
** trace_printk() being used. Allocating extra memory.  **
**                                                      **
** This means that this is a DEBUG kernel and it is     **
** unsafe for produciton use.                           **
**                                                      **
** If you see this message and you are not debugging    **
** the kernel, report this immediately to your vendor!  **
**                                                      **
**   NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE   **
**********************************************************`

teknoraver, 2016-05-29:
Source please, even to remove this big fat warning:

`**********************************************************
**   NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE   **
**                                                      **
** trace_printk() being used. Allocating extra memory.  **
**                                                      **
** This means that this is a DEBUG kernel and it is     **
** unsafe for produciton use.                           **
**                                                      **
** If you see this message and you are not debugging    **
** the kernel, report this immediately to your vendor!  **
**                                                      **
**   NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE   **
**********************************************************`

teknoraver, 2016-05-29:
Source please, even to remove this big fat warning:

```**********************************************************
**   NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE   **
**                                                      **
** trace_printk() being used. Allocating extra memory.  **
**                                                      **
** This means that this is a DEBUG kernel and it is     **
** unsafe for produciton use.                           **
**                                                      **
** If you see this message and you are not debugging    **
** the kernel, report this immediately to your vendor!  **
**                                                      **
**   NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE   **
**********************************************************```

blues-man, 2016-05-29:
How could we trust you if we can't see what you do? We love MIUI, and we trusted Lei Jun's words about it was worth to wait so long to get Mi5, but now that we finally hold it, we have a black box device since we can't look at the Android kernel, thus we cannot have the same freedom that let you give us a great Android device.

Fabio-San, 2016-05-29:
Come on, xiaomi, do this right favour to the devs ;)

anarkia1976, 2016-05-30:
Public pool? :100:

scafroglia93, 2016-05-30:
Sure ... Because of Xiaomi i bricked the phone. Stupid ootloader policy

proagg, 2016-05-30:
Come on Xiaomi, you have to release the kernel sources.

VidrikDev, 2016-06-02:
We all need kernel sources!

AayushJainRd7, 2016-06-07:
Kernel Sources!...cmon it's a snap device :-1:

anthyava, 2016-06-07:
Come on Xiaomi release  the kernel source

blues-man, 2016-06-10:
I don't think they'll ever read/care about this post, the only thing is vote massively on miui forum poll about releasing kernel source for mi5:
http://en.miui.com/forum.php?mod=viewthread&tid=277143

arter97, 2016-06-19:
Congratulations Xiaomi, you guys just made me to forcefully switch to OnePlus :D

anarkia1976, 2016-06-19:
The same, for this reason i have switched to wileyfox. ;)

Nimrodsv, 2016-06-22:
I was ---><--- close to buy MI5, saw no kernel source until now -> will buy other device

faust93, 2016-06-23:
Heh.. Have stepped into this shit once with buying "mipad one".. Xiaomi? Never again!

andy356, 2016-07-09:
Xiaomi, please release the kernel sources as required by law.

teknoraver, 2016-08-18:
release the sources NOW

BhSourav, 2016-08-25:
Source

Termynat0r, 2016-09-05:
I can't believe that this doesn't get prosecuted...

faust93, 2016-09-05:
Community should find out license advocates and sue xiaomi for GPL violations

teknoraver, 2016-09-05:
In China? lol

h2o64, 2016-09-08:
Xiaomi kept their engagement 5 months later.

closed 2016-09-08 by h2o64

akca, 2017-01-02:
Please update the kernel sources regularly. You just released once, but never updated. Please update the sources, respect your users Xiaomi! We want sources for Nougat!