MiCode/Xiaomi_Kernel_OpenSource #46: compile problem

eebssk1, 2016-05-19:
when i was compiling,i got this error

kernel/sysctl_binary.c:141:13: error: 'KERN_BOOT_REASON' undeclared here (not in a function)
kernel/sysctl_binary.c:528:13: error: 'NET_IPV6_ACCEPT_RA_PREFIX_ROUTE' undeclared here (not in a function)
kernel/sysctl_binary.c:528:2: error: initializer element is not constant
kernel/sysctl_binary.c:528:2: error: (near initialization for 'bin_net_ipv6_conf_var_table[24].ctl_name')
make[1]: *** [kernel/sysctl_binary.o] Error 1
make: *** [kernel] Error 2

kernel for redmi3/2
plesea help me

fedya, 2016-05-20:
Easy to fix, just drop this lines from code.
This lines removed years ago  as i remember.

eebssk1, 2016-05-20:
ok,but why don't they  update this file…

eebssk1, 2016-05-20:
drivers/char/diag/diagchar_core.c: In function 'diagchar_init':
drivers/char/diag/diagchar_core.c:2300:2: error: implicit declaration of function 'kmemleak_not_leak' [-Werror=implicit-function-declaration]
cc1: some warnings being treated as errors
make[3]: *** [drivers/char/diag/diagchar_core.o] Error 1
make[2]: *** [drivers/char/diag] Error 2
make[1]: *** [drivers/char] Error 2
make: *** [drivers] Error 2

help me…
@fedya

eebssk1, 2016-05-20:
all kmemleak_not_leak() can't be used

closed 2016-05-22 by eebssk1