MiCode/Xiaomi_Kernel_OpenSource #47: Compile error

cleverior, 2016-05-21:
I try to compile kernel using UBERTC 4.9 for my redmi3 but i stuck on this error

```
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  CALL    scripts/checksyscalls.sh
  CC      scripts/mod/devicetable-offsets.s
  GEN     scripts/mod/devicetable-offsets.h
  HOSTCC  scripts/mod/file2alias.o
  HOSTLD  scripts/mod/modpost
  CHK     include/generated/compile.h
  CHK     kernel/config_data.h
  CC      drivers/soc/qcom/subsystem_restart.o
drivers/soc/qcom/subsystem_restart.c: In function 'subsys_register':
drivers/soc/qcom/subsystem_restart.c:1480:19: error: 'WT_SUBSYSTEM_REASTART_LEVEL' undeclared (first use in this function)
  if (0 == strncmp(WT_SUBSYSTEM_REASTART_LEVEL, "SYSTEM", 6)) {
                   ^
drivers/soc/qcom/subsystem_restart.c:1480:19: note: each undeclared identifier is reported only once for each function it appears in
make[3]: *** [drivers/soc/qcom/subsystem_restart.o] Error 1
make[2]: *** [drivers/soc/qcom] Error 2
make[1]: *** [drivers/soc] Error 2
make: *** [drivers] Error 2
```

eebssk1, 2016-05-22:
you should make with this way.
make arch=arm64 cross_compile=you decide

eebssk1, 2016-05-22:
export doesn't work

cleverior, 2016-05-22:
Still stuck on the same error

eebssk1, 2016-05-22:
oh
ARCH
CROSS_COMPILE

eebssk1, 2016-05-22:
don' build it now 
the source code has errors

eebssk1, 2016-05-22:
forget my last comment.it can be built with a correct config

eebssk1, 2016-05-22:
sorry…i misunderstood you…
it seems the source code really having errors
i think this problem happens because it is missing header files.
@hehao
@MiCode

cleverior, 2016-05-23:
Waiting for update

eebssk1, 2016-05-23:
@cleverior
clever!?…
I found a way out!
in your .config
found CONFIG_MSM_SUBSYSTEM_RESTART=y
change to =n or just # drop it.
enjoy building
or you can still wait for update…😂

hehao, 2016-05-23:
Hi Guys,

You mentioned wrong person.

eebssk1, 2016-05-23:
same nickname"😱"

cleverior, 2016-05-23:
@eebssk1 : thanks for your info. I'll try it later.

cleverior, 2016-05-23:
stuck on new error 
```
make[1]: Nothing to be done for `dtbs'.
  CHK     include/generated/compile.h
  GZIP    kernel/config_data.gz
  CHK     kernel/config_data.h
  CC      drivers/video/msm/msm_fb.o
  CC      drivers/video/msm/mdp.o
  CC      drivers/video/msm/mdp_ppp.o
drivers/video/msm/msm_fb.c:30:24: fatal error: mach/board.h: No such file or directory
 #include <mach/board.h>
                        ^
compilation terminated.
make[3]: *** [drivers/video/msm/msm_fb.o] Error 1
make[3]: *** Waiting for unfinished jobs....
drivers/video/msm/mdp.c:28:27: fatal error: mach/hardware.h: No such file or directory
 #include <mach/hardware.h>
                           ^
compilation terminated.
make[3]: *** [drivers/video/msm/mdp.o] Error 1
drivers/video/msm/mdp_ppp.c:29:27: fatal error: mach/hardware.h: No such file or directory
 #include <mach/hardware.h>
                           ^
compilation terminated.
make[3]: *** [drivers/video/msm/mdp_ppp.o] Error 1
make[2]: *** [drivers/video/msm] Error 2
make[1]: *** [drivers/video] Error 2
make: *** [drivers] Error 2
```

eebssk1, 2016-05-23:
maybe it's gcc's problem
i use android ndk r10e
system environment's problem is possibe too…

webhaikal, 2016-05-26:
@cleverior, you need to resync the code. Mine builds fine but doesnt boot...

cleverior, 2016-05-28:
Which gcc u use ?

webhaikal, 2016-05-29:
4.9 UberTC

eebssk1, 2016-06-01:
compile finished! but touchscreen didn't work…

how about you?…

@cleverior

webhaikal, 2016-06-01:
How you got it to boot @eebssk1?

eebssk1, 2016-06-02:
@webhaikal
yes

cleverior, 2016-06-02:
Still stuck bro

webhaikal, 2016-06-02:
@eebssk1 How you compile it? What defconfig used and what changes?
@cleverior Same here...

eebssk1, 2016-06-02:
android r10e
wtbalabala-perf-defconfig
config_msm_subsystem_restart=n
silly batter states
touchscreen does't work
data doesn't work

so               the code is trash…

don't care about my english and i'm lazy………

eebssk1, 2016-06-02:
i use debian 8

eebssk1, 2016-06-02:
android ndk r10e………

cleverior, 2016-06-02:
Please share the link of gcc which you use.

eebssk1, 2016-06-03:
@cleverior 
32bit or 64bit system do you use?

cleverior, 2016-06-03:
64bit

webhaikal, 2016-06-03:
@cleverior @eebssk1 lets try to collaborate on this? Try to fix what we can and improve on it?

cleverior, 2016-06-03:
I still get error compile. Even i already try to use android ndk.

webhaikal, 2016-06-04:
Mine compiles but doesnt boot, using all known GCC (Sabermod, UberTC, AOSP)

cleverior, 2016-06-04:
Can we create a Hangout group ?

webhaikal, 2016-06-06:
@cleverior why not?

eebssk1, 2016-06-06:
which social group will you use?

webhaikal, 2016-06-06:
Why not do a Slack group? Can bring in rom devs too and such...

eebssk1, 2016-06-10:
temporarily fix your first error
add "int WT_SUBSYSTEM_REASTART_LEVEL" above if (0 == strncmp(WT_SUBSYSTEM_REASTART_LEVEL, "SYSTEM", 6)) {
@cleverior

cleverior, 2016-06-10:
how about this ?

 CC      fs/ext2/xattr.o
  CC      fs/ext2/xattr_user.o
  CC      fs/ext2/xattr_trusted.o
  LD      fs/ext2/ext2.o
  LD      fs/ext2/built-in.o
  CC      fs/ext3/balloc.o
In file included from fs/ext3/balloc.c:16:0:
fs/ext3/ext3.h:23:23: fatal error: linux/jbd.h: No such file or directory
 #include <linux/jbd.h>
                       ^
compilation terminated.
make[2]: *** [fs/ext3/balloc.o] Error 1
make[1]: *** [fs/ext3] Error 2
make: *** [fs] Error 2

eebssk1, 2016-06-10:
Don't have this problem…

vikoadi, 2016-06-12:
@eebssk1 I can build it using subsystem_restart off, but can't boot it. Adb show that it can't find any sound card
What I do is pulling boot.img from my phone, replace Image file using mkbootimg_tools, then reflash.

Are you rebuilding dt.img to get it to work?
I'm trying to but failed with 0 unique DTS found.

asiansaga, 2016-06-14:
premaca already doing great job on this, take a look at his code here [https://github.com/premaca/Xiaomi_Kernel_OpenSource/tree/lp5.x](url)

i can compile using his defconfig, and get fix on sound and touchscreen, there are still wifi error left, and some sensors i think.

count me in if you guys want to create some groups

eebssk1, 2016-06-14:
404
@asiansaga

asiansaga, 2016-06-14:
i've edited the link

@eebssk1

cleverior, 2016-06-15:
oh maaaan... so many errors i found with this source kernel So confusing. -_-

And i already try to use source from premaca but i get this error 

  CC      scripts/mod/empty.o
  HOSTCC  scripts/selinux/genheaders/genheaders
gcc: error: unrecognized command line option ‘-mlittle-endian’
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
make[2]: *** [scripts/mod/empty.o] Error 1
make[1]: *** [scripts/mod] Error 2
make[1]: *** Waiting for unfinished jobs....
  HOSTCC  scripts/selinux/mdp/mdp
make: *** [scripts] Error 2
make: INTERNAL: Exiting with 5 jobserver tokens available; should be 4!

any ide @asiansaga ?

asiansaga, 2016-06-15:
@cleverior 
I don't meet those error, are you sure using the right defconfig, also try to run make distclean before running another make

webhaikal, 2016-06-15:
@asiansaga You can get it to boot? I can't seem to get it to boot even using that repo...

asiansaga, 2016-06-15:
@webhaikal yup, boot but without WiFi, and some sensor not working. Build from this repo, with some adjustments from @premaca repo

webhaikal, 2016-06-15:
@asiansaga I can't seem to get it to boot... I took the changes from @premaca and combined it with this but still doesnt work. What steps you did, what toolchain used and such?

asiansaga, 2016-06-16:
@webhaikal 
Let me tell you from the start 😁
1. Build from this repo, with wt88047-perf_defconfig, found error with MSM_SUBSYSTEM_RESTART
2. Googling here, got suggestions from @eebssk1 to remark it in .config file, got it build and also boot with no touchscreen 
3. Ask to Miui forum and directed to @premaca repo to fix touchscreen, build with no simcard support 
4. Look deeper to premaca repo, found difference in file drivers/soc/qcom/subsytem_restart.c, put it in my working source, build and got no sound 
5. There's a fix for sound in premaca repo, build and no wifi
6. Haven't got spare time to look how to fix it, I just wait for master @premaca to fix it 😄

I use UBERTC 4.9 to build and android kitchen from osm0sis @ xda to unpack - repack boot.img 

Flash to device using fastboot

webhaikal, 2016-06-16:
@asiansaga is it the wt88047-perf or the wt88509-64-perf?

eebssk1, 2016-06-16:
@asiansaga
you can manually download the wifi driver source code  from codeaurora,
BUT I got error when i was buiding it。。。

asiansaga, 2016-06-16:
@webhaikal wt88047-perf and make sure the arch=arm not arm64 cause MIUI still build using ARM architecture

@eebssk1 any link there?

asiansaga, 2016-06-16:
@webhaikal i forgot that my device is redmi 2 prime, sorry if you think it's a redmi3

WinKarbik, 2016-07-13:
I ll try fix kernel source for redmi 3

rushabh71, 2016-07-26:
Anyone able to make dt.img for xiaomi mi4i kernel?
giving me an error..

`DTB combiner:
Input directory: '/home/rushabh/KernelDevelopment/Xiaomi_Kernel_OpenSource-ferrari-l-oss/arch/arm64/boot/dts/'

Output file: '/home/rushabh/KernelDevelopment/Xiaomi_Kernel_OpenSource-ferrari-l-oss/arch/arm64/boot/dt.img'

Searching subdir: /home/rushabh/KernelDevelopment/Xiaomi_Kernel_OpenSource-ferrari-l-oss/arch/arm64/boot/dts/include/ ... 

=> Found 0 unique DTB(s)`

as i have read various threads but can not able to fix this. i have used dtbToolCm and dtbtool (version -3 and -2) also same result.

chhapil, 2016-07-27:
your device tree files seems to be located in
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/ferrari-l-oss/arch/arm/boot/dts/qcom
try changing it and see if it can generate the dtbs

rushabh71, 2016-07-27:
@chhapil tried it, but still not able to find dtb

`./dtbToolCM -2 -s 2048 -o /home/rushabh/Desktop/dt.img -p /home/rushabh/KernelDevelopment/android_kernel_xiaomi_ferrari/scripts/dtc/ /home/rushabh/KernelDevelopment/android_kernel_xiaomi_ferrari/arch/arm/boot/dts/qcom/
DTB combiner:
  Input directory: '/home/rushabh/KernelDevelopment/android_kernel_xiaomi_ferrari/arch/arm/boot/dts/qcom/'
  Output file: '/home/rushabh/Desktop/dt.img'
=> Found 0 unique DTB(s)`

If i change input directory like below than it works but it is using 'arm' dtb, i think so....

./dtbToolCM -2 -s 2048 -o /home/rushabh/Desktop/dt.img -p /home/rushabh/KernelDevelopment/android_kernel_xiaomi_ferrari/scripts/dtc/ /home/rushabh/KernelDevelopment/android_kernel_xiaomi_ferrari/arch/arm/boot/
DTB combiner:
  Input directory: '/home/rushabh/KernelDevelopment/android_kernel_xiaomi_ferrari/arch/arm/boot/'
  Output file: '/home/rushabh/Desktop/dt.img'
Found file: msm8936-qrd-skuk.dtb ... 
Version:2
chipset: 233, rev: 0, platform: 65547, subtype: 5, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
additional chipset: 240, rev: 0, platform: 65547, subtype: 5, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
additional chipset: 242, rev: 0, platform: 65547, subtype: 5, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
Searching subdir: /home/rushabh/KernelDevelopment/android_kernel_xiaomi_ferrari/arch/arm/boot/bootp/ ... 
Found file: msm8939-mtp-jdi.dtb ... 
Version:2
chipset: 239, rev: 0, platform: 8, subtype: 3, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
additional chipset: 241, rev: 0, platform: 8, subtype: 3, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
additional chipset: 263, rev: 0, platform: 8, subtype: 3, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
Found file: msm8916-mtp-smb1360.dtb ... 
Version:2
chipset: 206, rev: 0, platform: 8, subtype: 1, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
additional chipset: 248, rev: 0, platform: 8, subtype: 1, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
additional chipset: 249, rev: 0, platform: 8, subtype: 1, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
additional chipset: 250, rev: 0, platform: 8, subtype: 1, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
Found file: msm8936-cdp.dtb ... 
Version:2
chipset: 233, rev: 0, platform: 1, subtype: 0, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
additional chipset: 240, rev: 0, platform: 1, subtype: 0, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
additional chipset: 242, rev: 0, platform: 1, subtype: 0, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
Found file: msmtellurium-sim.dtb ... 
Version:2
chipset: 264, rev: 0, platform: 16, subtype: 0, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
Searching subdir: /home/rushabh/KernelDevelopment/android_kernel_xiaomi_ferrari/arch/arm/boot/compressed/ ... 
Found file: msm8939-mtp-smb1360.dtb ... 
Version:2
chipset: 239, rev: 0, platform: 8, subtype: 0, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
additional chipset: 241, rev: 0, platform: 8, subtype: 0, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
additional chipset: 263, rev: 0, platform: 8, subtype: 0, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
Found file: msm8939-cdp-jdi.dtb ... 
Version:2
chipset: 239, rev: 0, platform: 1, subtype: 1, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
additional chipset: 241, rev: 0, platform: 1, subtype: 1, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
additional chipset: 263, rev: 0, platform: 1, subtype: 1, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
Found file: msm8916-mtp.dtb ... 
Version:2
chipset: 206, rev: 0, platform: 8, subtype: 0, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
additional chipset: 248, rev: 0, platform: 8, subtype: 0, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
additional chipset: 249, rev: 0, platform: 8, subtype: 0, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
additional chipset: 250, rev: 0, platform: 8, subtype: 0, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
Found file: msm8939-sim.dtb ... 
Version:2
chipset: 239, rev: 0, platform: 16, subtype: 0, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
additional chipset: 241, rev: 0, platform: 16, subtype: 0, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
additional chipset: 263, rev: 0, platform: 16, subtype: 0, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
Found file: msm8936-mtp.dtb ... 
Version:2
chipset: 233, rev: 0, platform: 8, subtype: 1, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
additional chipset: 240, rev: 0, platform: 8, subtype: 1, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
additional chipset: 242, rev: 0, platform: 8, subtype: 1, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
Found file: msmtellurium-rumi.dtb ... 
Version:2
chipset: 264, rev: 0, platform: 15, subtype: 0, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
Found file: msm8939-cdp.dtb ... 
[... 3 lines cut ...]
additional chipset: 263, rev: 0, platform: 1, subtype: 0, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
Searching subdir: /home/rushabh/KernelDevelopment/android_kernel_xiaomi_ferrari/arch/arm/boot/dts/ ... 
Searching subdir: /home/rushabh/KernelDevelopment/android_kernel_xiaomi_ferrari/arch/arm/boot/dts/include/ ... 
Searching subdir: /home/rushabh/KernelDevelopment/android_kernel_xiaomi_ferrari/arch/arm/boot/dts/qcom/ ... 
Searching subdir: /home/rushabh/KernelDevelopment/android_kernel_xiaomi_ferrari/arch/arm/boot/dts/testcases/ ... 
Found file: msm8939-mtp.dtb ... 
Version:2
chipset: 239, rev: 0, platform: 8, subtype: 1, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
additional chipset: 241, rev: 0, platform: 8, subtype: 1, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
additional chipset: 263, rev: 0, platform: 8, subtype: 1, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
[... cut ...]
additional chipset: 242, rev: 0, platform: 1, subtype: 1, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
Found file: msm8936-mtp-smb1360.dtb ... 
Version:2
chipset: 233, rev: 0, platform: 8, subtype: 0, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
additional chipset: 240, rev: 0, platform: 8, subtype: 0, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
additional chipset: 242, rev: 0, platform: 8, subtype: 0, pmic0: 0, pmic1: 0, pmic2: 0, pmic3: 0
=> Found 43 unique DTB(s)

Generating master DTB... completed

is it ok to use dtb of /arm/boot/ ?

chhapil, 2016-07-27:
you are compiling for Mi4i right? thats MSM8939
Using arm directory should be fine to generate the dtb, nothing to do with
32 bit or 64 with it

chhapil, 2016-07-27:
another hint, just decompile your boot.img and validate the dtb file size
from it, whether its using a combined dtb or only specific one.
I had this experience while compiling kernel for Lenevo A6000, it was using
combined dtb.

Cheers

rushabh71, 2016-07-27:
@chhapil thanks man. will try to boot with this generated dt.img.

ma3axaka007, 2016-08-22:
I think it should be ```#define WT_SUBSYSTEM_REASTART_LEVEL "related"``` in drivers/soc/qcom/subsystem_restart.c on line 15, as in [this](https://github.com/yantz74/Ido_Redmi3_Kernel/blob/ido-l-oss/drivers/soc/qcom/subsystem_restart.c) fork