MiCode/Xiaomi_Kernel_OpenSource #48: compiling error

eebssk1, 2016-05-22:
In file included from include/linux/kernel_stat.h:8:0,
                 from init/main.c:32:
include/linux/interrupt.h: In function ‘tasklet_unlock’:
include/linux/interrupt.h:500:2: error: implicit declaration of function ‘smp_mb__before_atomic’ [-Werror=implicit-function-declaration]
  smp_mb__before_atomic();
  ^
include/linux/interrupt.h: In function ‘tasklet_disable_nosync’:
include/linux/interrupt.h:548:2: error: implicit declaration of function ‘smp_mb__after_atomic’ [-Werror=implicit-function-declaration]
  smp_mb__after_atomic();
  ^
cc1: some warnings being treated as errors
scripts/Makefile.build:308: recipe for target 'init/main.o' failed
make[1]: *** [init/main.o] Error 1
Makefile:822: recipe for target 'init' failed
make: *** [init] Error 2

too many errors.....
ido-l-oss
@hehao

eebssk1, 2016-05-22:
fixed

eebssk1, 2016-05-22:
another erroe
drivers/soc/qcom/subsystem_restart.c: In function 'subsys_register':
drivers/soc/qcom/subsystem_restart.c:1480:19: error: 'WT_SUBSYSTEM_REASTART_LEVEL' undeclared (first use in this function)
  if (0 == strncmp(WT_SUBSYSTEM_REASTART_LEVEL, "SYSTEM", 6)) {
                   ^
drivers/soc/qcom/subsystem_restart.c:1480:19: note: each undeclared identifier is reported only once for each function it appears in
scripts/Makefile.build:308: recipe for target 'drivers/soc/qcom/subsystem_restart.o' failed
make[3]: *** [drivers/soc/qcom/subsystem_restart.o] Error 1
make[3]: *** 正在等待未完成的任务....
scripts/Makefile.build:455: recipe for target 'drivers/soc/qcom' failed
make[2]: *** [drivers/soc/qcom] Error 2
scripts/Makefile.build:455: recipe for target 'drivers/soc' failed
make[1]: *** [drivers/soc] Error 2
make[1]: *** 正在等待未完成的任务....
  LD      drivers/video/msm/mdss/built-in.o
  LD      drivers/video/msm/built-in.o
  LD      drivers/video/built-in.o
Makefile:822: recipe for target 'drivers' failed
make: *** [drivers] Error 2

it seems.missing header files
@MiCode
@hehao

cleverior, 2016-05-23:
Same with me

eebssk1, 2016-05-23:
@cleverior
clever!?…
I found a way out!
in your .config
found CONFIG_MSM_SUBSYSTEM_RESTART=y
change to =n or just # drop it.
enjoy building~

closed 2016-05-23 by eebssk1

ashoksoni, 2016-09-15:
@eebssk1 

How did you fix your first issue?

tongo45, 2016-12-23:
@eebssk1 
how to fix first issue

eebssk1, 2017-02-11:
@tongo45 Ehh....This is fixed in the new source..Try git pull.

bspteam, 2019-05-13:
> fixed

@eebssk1  how to fix the first issue

eebssk1, 2019-05-21:
For anyone who still gets stuck because of this issue, pls go clone a CAF base kernel.Then apply device-specified patches(If you have enough experience, you can go messing with comments in this repo, otherwise, go get some help from XDA.)