MiCode/Xiaomi_Kernel_OpenSource #53: You add Xiaomi as copyright holder unnecessary

kumajaya, 2016-06-22:
You add Xiaomi as copyright holder some Linux Foundation files unnecessary, one example https://github.com/MiCode/Xiaomi_Kernel_OpenSource/blob/ido-l-oss/drivers/char/diag/diag_memorydevice.c#L2 That file identical to upstream, tell me in which part from Xiaomi and/or Wingtech.

Anik1199, 2016-06-22:
Haha

kumajaya, 2016-06-23:
@xnickbrandtx I want he stop it, also found same issue inside libra/aqua source code. This is a complaint from open source community. Add a single line " * Copyright (C) 2016 XiaoMi, Inc." is a shameful.

xnickbrandtx, 2016-06-23:
@kumajaya I wish I could give you internet/power now so that you can upload the patch. I want to try the new boot loader hehe :D

ramsudharsan, 2016-06-24:
And I discovered a file where Xiaomi specifying wrong battery capacity (https://github.com/MiCode/Xiaomi_Kernel_OpenSource/blob/ido-l-oss/arch/arm/boot/dts/qcom/batterydata-wingtech-feimaotui_SONY-4v4-4000mah.dtsi#L18)
Compare the battery capacity given in the file name and the one specified as qcom parameter in device and I guess either one of them must be wrong.

xnickbrandtx, 2016-06-25:
@ramsudharsan well xiaomi always had a problem with math. They stated the battery I have either contained 2200mAh or 2250mAh

kumajaya, 2016-06-25:
@hehao If you want a fair interchange, remove XiaoMi copyright string from driver/char/diag and I will tell you code bugs in smb1360-charger-fg and bq2022a.

jark147, 2016-09-08:
@kumajaya  have you good idea in smb1360-charger-fg and bq2022a, and what mistakes for it?

jark147, 2016-09-08:
some debug log will be remove and it's useless,  i think that copyright issue will be give attention for next time

watbulb, 2020-08-18:
Old issue, but I've brought this up with GitHub support staff as this garbage continues today.