MiCode/Xiaomi_Kernel_OpenSource #54: redmi 3s (land) kernel source

zwliew, 2016-06-24:
nice device; mind uploading the kernel source asap? thanks :)

hewz, 2016-08-02:
+10000

utopio, 2016-08-07:
Yes
It's necessary.

LKseven, 2016-08-15:
source source source

bullik01, 2016-08-22:
Yes. We need kernel sources 4 land.

jesuscea, 2016-08-30:
I agree too, we need kernel sources for Redmi 3S please :+1:

hewz, 2016-08-31:
miui sucks.. we need kernel files for land

LiberteCzech, 2016-09-04:
Please observe the conditions of the GPL and release resources.
There will also be a greater chance for the customer buys.

Whatitat1930, 2016-09-05:
This needs to happen, come on Xiaomi! If this kernel sources, it will be flying off the shelves

sheshuv, 2016-09-06:
Kernel source please?

Fairsti, 2016-09-06:
Plz release kernel sources!

ZelectroKarZ, 2016-09-07:
Kernel sources so that we can build CM and aospa for Land/3S Prime please!

IITiti92II, 2016-09-07:
Xiaomi please release kernel source!

rigaz29, 2016-09-07:
https://wiki.codeaurora.org/xwiki/bin/QAEP/release msm8937_64

Silentlys, 2016-09-09:
+n+1

Silentlys, 2016-09-09:
https://github.com/Silentlys/android_device_xiaomi_land

bishnuprasad884uu, 2016-09-10:
Yes... We need the source

rodsayd, 2016-09-10:
Please the sourcee

dan-man, 2016-09-14:
need this

skrz23, 2016-09-15:
the sources

xdarknet, 2016-09-15:
Hope the kernel source release this month

ZelectroKarZ, 2016-09-23:
I think cm is starting to build for this device,,, They're waiting for the kernel sources... Found this on CodeAurora: https://wiki.codeaurora.org/xwiki/bin/QAEP/release

Search msm8937_64 on the page.

busote, 2016-09-27:
Yes. We need kernel sources 4 land.

crissdua, 2016-09-28:
We need kernel please

ttkea, 2016-10-05:
请开源，谢谢

supercairos, 2016-10-06:
I tried to do my own, but I seams to have some minor issues. 
https://github.com/supercairos/android_kernel_xiaomi_msm8937/

np20feb, 2016-10-09:
+1 ...asap

wangyangdahai, 2016-10-10:
**redmi 3X** kernel source **!!!!!**

Lautta17, 2016-10-16:
+1... Please

diparthshah, 2016-10-17:
+∞

hazehunter, 2016-10-18:
please release it ASAP so we can have further development of this lovely device

LiberteCzech, 2016-10-21:
Xiaomi observe the open GPL license!

ttkea, 2016-10-22:
pls

0PED1A, 2016-10-23:
Nothing yet...!! :-(

charansai612, 2016-10-23:
Release It AsAp :)

DroidThug, 2016-10-23:
LA Xiaomi! Do it fast, ya Thug!

Or do you wanta formal GPL Notice now?

ptmaroct, 2016-10-24:
Do it boi

mkreja, 2016-10-24:
+1
yes!

SayanAcharya, 2016-10-25:
Pls release the sources for further development purposes...

phoenixprojects, 2016-10-26:
Xiaomi pls release the kernel sources. Enough already

DroidThug, 2016-10-27:
Anyone who'd like to help me in achieving an EFF backing?

LiberteCzech, 2016-10-27:
[](https://www.change.org/p/publish-the-kernel-source-for-the-xiaomi-redmi-3s-prime-3x?recruiter=601727891&utm_source=share_petition&utm_medium=copylink)
**Petition for the release of the kernel source of the Redmi 3S/Prime/3X**

harrzz, 2016-10-30:
Yes, please release the kernel sources for the Redmi 3s

INCTeam8, 2016-11-05:
We need the kernel source for the Redmi 3s,please release it!

mohdmujahidansari526, 2016-11-05:
please release the Kernel Source for Redmi 3S Prime.

Nekit10, 2016-11-06:
We need source pls!!!

k3a, 2016-11-06:
Yes, I like the device but won't buy it until kernel sources are released!

DenHafiz69, 2016-11-15:
Please release it so developers can build CM rom :+1:

k3a, 2016-11-18:
He Hao (何浩) said it has been included in the plan, and plan to open Redmi 3S kernel source in next month. Fingers crossed. :)

utopio, 2016-11-18:
wow
that's great.

np20feb, 2016-11-18:
@k3a...hope they hold true to their words !!

DeckardShaw, 2016-11-18:
@k3a 
from  where you got the info?
source? :p

k3a, 2016-11-18:
Source: I asked him by email and he responded, so I shared it with others.

jagdishphadke, 2016-11-19:
Lets hope that xiaomi will do it soon

legion111, 2016-11-19:
3s

legion111, 2016-11-19:
Hope they will release kernel source of redmi 3s prime soon

utopio, 2016-11-19:
I think miui won't let us down.

DroidThug, 2016-11-19:
Nice news! Would be great!

legion111, 2016-11-20:
Any Proof k3a ?? How Can We Believe That They Are Releasing Source Next Month

legion111, 2016-11-20:
Please send mail screenshot For The Satisfaction Of Redmi 3s Prime Owners :) waiting for your reply thanks....

DeckardShaw, 2016-11-20:
Ya scrnsht will be nice for satisfaction :D

[35 more replies]

closed 2017-01-02 by zwliew