MiCode/Xiaomi_Kernel_OpenSource #58: net/built-in.o

rushabh71, 2016-07-25:
CC      net/xfrm/xfrm_user.o
  CC      net/wireless/ibss.o
  CC      net/sysctl_net.o
  CC      net/xfrm/xfrm_ipcomp.o
  CC      net/wireless/sme.o
  LD      net/xfrm/built-in.o
  CC      net/wireless/chan.o
  CC      net/wireless/ethtool.o
  CC      net/wireless/mesh.o
  CC      net/activity_stats.o
  CC      net/wireless/ap.o
  CC      net/wireless/trace.o
  CC      net/wireless/wext-core.o
  CC      net/wireless/wext-proc.o
  CC      net/wireless/wext-spy.o
  CC      net/wireless/wext-priv.o
  CC      net/wireless/regdb.o
  LD      net/wireless/cfg80211.o
  LD      net/wireless/built-in.o
  LD      net/built-in.o
rushabh@rushabh-Lenovo-G580:~/KernelDevelopment/Xiaomi_Kernel_OpenSource-ferrari-l-oss$ 

compilation stops here...
My system:
toolchain- NDK version 4.9
device - Mi4i
Ubuntu 64 bit

ramsudharsan, 2016-07-25:
scroll above in your terminal to see the exact error; 
net/built-in.o is not the correct title for this issue;

rushabh71, 2016-07-25:
yup solved that i was using multiple threads. 

please can you look here this error..

http://forum.xda-developers.com/mi-4i/help/kernel-compilation-error-t3425748

thank you...

ramsudharsan, 2016-07-25:
just do a clean build;

closed 2016-07-25 by rushabh71