MiCode/Xiaomi_Kernel_OpenSource #60: Incomplete Source HM Note 3 aka kenzo

khanfrd, 2016-08-20:
The **new Redmi Note 3** uses **goodix fingerprint drivers**, and the source is outdated
**so please update the source ASAP**

alesasnouski, 2016-08-22:
Guys, can you update source code ?

Alex0090, 2016-08-23:
Confirm, please update the source

amulecregg, 2016-08-25:
Update the source code.

arhue, 2016-08-25:
This needs urgent attention. Changing hardware has broken certain things on AOSP based ROMs. I think some people are reporting VoLTE and front camera not working?

roastedwildebeest, 2016-08-25:
Please update the kernel source. Fingerprint and Front Camera are not working.

Tyaginator, 2016-08-25:
Update update update update update!

Janikeus, 2016-08-25:
Hi guys, updating the source code would be great.
Since new type of fingerprint sensor and front camera are not working in AOSP based ROM. :)

acnalogia, 2016-08-25:
Please update the kernel source.
fingerprint sensor and front camera are not working in AOSP based ROM

Maria-gr, 2016-08-25:
Update the sources please !! Fingerprint and front camera not working on newer models..

arhue, 2016-08-25:
Actually it's not that Xiaomi can choose to not update the sources. They are forced to since the kernel uses GPLv2 and they already have distributed the updated kernel to the users. They are in India so they can be sued for this.

gingerboy92, 2016-08-26:
Update the source code please!!!

khanfrd, 2016-08-28:
It's been week no reply from xiaomi, that what I expected 
anyway I got in touch with yijiangtao author of our goodix driver, he say the the current code of goodix we have right now are fine they should work, the issue is with DTS or maybe hall,
I will take a look again, it will be nice if other dev take a look too

alesasnouski, 2016-08-28:
What abount front camera ?

khanfrd, 2016-08-28:
I have the newer hardware one and don't have any issue with cam on miui

yemjalakumar, 2016-09-03:
Update the source code please !

Whitexp, 2016-09-03:
xiaomi , we waiting for u update the source .

tank0412, 2016-09-04:
Hello, guys.
I am a rom developer. Kernel sources by Xiaomi has **goodix and front camera driver**.
**Again. Sources by Xiomi is COMPLETE.**
So you may ask me why it does not work on custom rom.
It does not work because **developer of official CM removed goodix driver from kernel sources.**
Also he removed all files for goodix and fingerprint in rom.
So it is not a fault of Xiaomi. It is a fault of TheStrix.

P.S. I fixed front camera issue on officail CM builds and published fix. Are you still do not know about it???

alesasnouski, 2016-09-04:
Can you share your PR with us? Thank you.

tank0412, 2016-09-04:
http://forum.xda-developers.com/showpost.php?p=68183787&postcount=18

khanfrd, 2016-09-04:
@tank0412 first of all its no one fault not all the user have goodix fp,
and some user reported goodix fp doesn't work on miui too i don't know exactly as i have fpc1020

and yes i am too still confused with TheStrix, i dont know why he use MM blobs, as much i know about roms we can use lp blobs too

and as you say you are rom developer can you help me out with something???

and if any user with goodix can confirm and clean this misunderstanding about goodix fp on miui it will be great help

alex9yust, 2016-09-04:
@tank0412 how run gx_fpd? in miui sets fingerprintd and gx_fpd, and fingerprintd from cyanogenmod knows nothing about gx_fpd.
it certainly does not belong to the kernel.

arhue, 2016-09-05:
"and if any user with goodix can confirm and clean this misunderstanding about goodix fp on miui it will be great help
if goodix with custom kernel works on miui then there is no issue with source,"
I think you should wait for some senior developer to confirm. Regardless if Xiaomi had updated the kernel during the last 5 months after they released the original sources, they have to update the sources to match the released kernel version technically. But I think we can give them a pass on it. Just wait for someone to confirm to sort the issue out.

poad42, 2016-09-05:
@tank0412 Buddy if you are a ROM developer you'd know my taking a look at cm sources even the FPC driver used is not from xiaomi's sources but knaged from another device. To even make use of xiaomi's sources they'd have to open source the fingerprint api that they used in miui lollipop, which they won't obviously. So we cannot use the existing goodix and FPC drivers because they are all outdated. Regarding the camera sensor, xiaomi's has only sourced one of them while using two types of sensor for their units same goes to gyroscope as well. Therefore they have to update their and its not complete by a country mile.

tank0412, 2016-09-05:
@khanfrd I have phone with goodix fingerprint. And it works on MIUI MM **sometimes** (but it works)
We can use lollipop blobs **BUT** we need libshims for it and AFAIK RIL does not work on Lollipop blobs. (correct if i wrong)
Even if use LP blobs, fingerprint will not work due to major differentces in fingerprint MM API.

khanfrd, 2016-09-05:
@tank0412 what about lp?? goodix work with custom kernel on lp??

GuneetAtwal, 2016-09-05:
goodix doesnt work because it is broken! i  and @spbnick have tried almost every method to fix goodix fp and we have got to making it appear in the fp settings.

TheStrix removed it because they were not able to fix the issue. Santosh posted his goodix fix but never open sourced it due to which we have to face so many problems.

tank0412, 2016-09-06:
@GuneetAtwal But Santosh M just port kernel from Miui MM to CM13. Nothing else.
So it can not be opensourced.

GuneetAtwal, 2016-09-06:
nope, he made a patch for goodix sensor , also his device trees are closed sourced

nicknitewolf, 2016-09-06:
@GuneetAtwal true. He does not open the source. Even Xiaomi updated their own internal sources of L yet they never release.

arhue, 2016-09-06:
@GuneetAtwal Is it some kind of a shim that he has made for the FP API?
He is not forced to open source it  thanks to the permissive Apache license but we can pressure him to.

GuneetAtwal, 2016-09-06:
why pressurise him ? let him do whatever he wants to do. If he doesn't care about us we also dgaf.

abhis3k, 2016-09-09:
And now Global Beta and China Beta for MM is out, and we do not have yet the kernel sources.
We need the sources to be updated to MM with all sensors, so custom rom/kernel support can complete

Altairko, 2016-09-26:
Please update the kernel source. Fingerprint and Front Camera are not working.

sumchamp, 2016-09-26:
Xiaomi, please also release MM sources for VoLTE.

tmaone, 2016-09-29:
http://gpl-violations.org/

khanfrd, 2016-09-29:
Now we know source is complete for Lollipop
And official Marshmallow Rom is yet to come, So we cant bug Xiaomi yet
So closing the issue

LightTube, 2017-01-17:
MM realeased now, open it

yemjalakumar, 2017-01-18:
Official global beta mm  released and it is stable too

poad42, 2017-01-18:
Drop it guys, they won't give us an updated kernel source. Knowing xiaomi's history of GPL violations, they are never going to update the source.

khanfrd, 2017-01-18:
As we know [official marshmallow](http://en.miui.com/download-301.html) is up and according to gpl kernel source need to update too
So Xiaomi please update fucking kernel source too!!!!!!!!!!!!

AmolAmrit, 2017-04-07:
Even If many things got fixed now. We still have the old Kernel sources. So please Xiaomi release kernel sources.

2272995821, 2018-01-11:
Plz update it