MiCode/Xiaomi_Kernel_OpenSource #69: redmi 3s (land) kernel source please

sheshuv, 2016-09-06:
Please release redmi 3s prime Kernel source

rigaz29, 2016-09-07:
https://wiki.codeaurora.org/xwiki/bin/QAEP/release msm8937_64

m0eter, 2016-10-21:
Rigaz, stop posting this: it's not a kernel for the redmi 3s, just for some dev boards with our SoC.

zwliew, 2016-10-30:
C'mon Xiaomi :)

ramsudharsan, 2016-10-31:
I worked a few stuffs for the past few hours for the Xiaomi Redmi 3S Prime:
1. Extracted individual dts files from kernel (todo: have to study individual files)
2. Merged linux kernel changes (upto 3.18.44) by taking a caf tag as base
3. Merged f2fs for 3.18.y branch from kernel.org
4. Did a test build of the kernel to make sure everything merged so far compiles without issues

P.S. Some of the above mentioned work could have been avoided if Xiaomi had released the kernel sources,

psndna88, 2016-10-31:
get matching prima wifi source from caf aswell
i made my Android mm & N kernel for kenzo similarly...

INCTeam8, 2016-11-05:
C'mon xiaomi we need the kernel source

Nekit10, 2016-11-16:
We need source pls!

pavan3999, 2016-11-19:
please release kernel source

ljmf00, 2016-11-27:
we need the kernel source

ljmf00, 2020-09-26:
This may be closed.