MiCode/Xiaomi_Kernel_OpenSource #70: pn548.h missing

teknoraver, 2016-09-11:
Compilation error with Mi5 source code and gemini_user_defconfig

```  CC      drivers/nfc/pn548/pn548.o
drivers/nfc/pn548/pn548.c:43:19: fatal error: pn548.h: No such file or directory
 #include <pn548.h>
                   ^
compilation terminated.
scripts/Makefile.build:257: recipe for target 'drivers/nfc/pn548/pn548.o' failed
make[3]: *** [drivers/nfc/pn548/pn548.o] Error 1
scripts/Makefile.build:402: recipe for target 'drivers/nfc/pn548' failed
make[2]: *** [drivers/nfc/pn548] Error 2
scripts/Makefile.build:402: recipe for target 'drivers/nfc' failed
make[1]: *** [drivers/nfc] Error 2
Makefile:945: recipe for target 'drivers' failed
make: *** [drivers] Error 2```

h2o64, 2016-09-11:
"OH! A wild pn548.h appears !!! Catch them all !" https://github.com/MiCode/Xiaomi_Kernel_OpenSource/blob/gemini-m-oss/drivers/nfc/pn548/pn548.h

(TIP: Replace < > with " " at https://github.com/MiCode/Xiaomi_Kernel_OpenSource/blob/gemini-m-oss/drivers/nfc/pn548/pn548.c#L43

teknoraver, 2016-09-11:
yes I know, but they should do it, not me

h2o64, 2016-09-11:
This kernel is made to be built wih ABS (Android Build System) and not isolated.
They pushed a kernel which works with ABS.
ABS has a specific kernel header managment and that's why your facing this error (and #71 too) so they don't have to be responsable of it

closed 2024-01-19 by teknoraver