MiCode/Xiaomi_Kernel_OpenSource #71: wrong path for tracer_pkt_private.h

teknoraver, 2016-09-11:
While building Mi5 sources

```
  CC      drivers/soc/qcom/tracer_pkt.o
In file included from drivers/soc/qcom/tracer_pkt_private.h:49:0,
                 from drivers/soc/qcom/tracer_pkt.c:18:
include/trace/define_trace.h:83:43: fatal error: ./tracer_pkt_private.h: No such file or directory
 #include TRACE_INCLUDE(TRACE_INCLUDE_FILE)
                                           ^
compilation terminated.
scripts/Makefile.build:257: recipe for target 'drivers/soc/qcom/tracer_pkt.o' failed
make[3]: *** [drivers/soc/qcom/tracer_pkt.o] Error 1
scripts/Makefile.build:402: recipe for target 'drivers/soc/qcom' failed
make[2]: *** [drivers/soc/qcom] Error 2
scripts/Makefile.build:402: recipe for target 'drivers/soc' failed
make[1]: *** [drivers/soc] Error 2
Makefile:945: recipe for target 'drivers' failed
make: *** [drivers] Error 2
make: Leaving directory '/home/matteo/src/xiaomi/kernel'
```

RiTZ-AFR, 2016-12-06:
in tracer_pkt_private.h 

#undef TRACE_INCLUDE_PATH
#define TRACE_INCLUDE_PATH . <--remove this dot                                 
#undef TRACE_INCLUDE_FILE 
#define TRACE_INCLUDE_FILE (insert your folder path here)tracer_pkt_private

here's what i've done

#undef TRACE_INCLUDE_PATH
#define TRACE_INCLUDE_PATH 
#undef TRACE_INCLUDE_FILE 
#define TRACE_INCLUDE_FILE home/ritz/android/Xiaomi_Kernel_OpenSource-gemini-m-oss/drivers/soc/qcom/tracer_pkt_private

hope this help

Uvneshkumar, 2018-01-31:
Thank you so much. Got the same kind of error. Now resolved using your idea. Thanks a lot.

linde12, 2018-08-02:
I'm getting multiple similar errors when trying to build for Mi MIX. Is the source outdated or are we missing something (like some special toolchain)? It's not very difficult to fix, but i wish Xiaomi would just disclose their _actual_ source code as per GPL...

gxr441, 2018-08-02:
Oh I thought I was doing something wrong. I was getting file not found and many forbidden warnings. Though I built the kernel for Vince, correcting all file paths and disabling warnings, the kernel didn't even boot.

Morrigan-Ship, 2025-05-24:
help me same error

teknoraver, 2025-09-25:
I changed that phone 7 years ago, lol