MiCode/Xiaomi_Kernel_OpenSource #72: firmware files missing

teknoraver, 2016-09-11:
To correctly build the kernel you need this two files in firmware

[firmware.zip](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/files/466127/firmware.zip)

bgcngm, 2016-09-11:
Where did you get those from?

teknoraver, 2016-09-11:
just `objcopy -I binary -O ihex IN.fw OUT.fw.ihex`

bgcngm, 2016-09-11:
As I said, I  think these are not needed for any gemini variant. Did you manage to get a bootable kernel?

teknoraver, 2016-09-11:
Not yet. My compiled kernel gets 8 MB smaller, I noticed that in the Xiaomi one there are 8 MB of DeviceTree data appended.
I tried to append it to my kernel but it won't boot

chhapil, 2016-09-12:
you would need combined DTB I guess

bgcngm, 2016-09-12:
@teknoraver, are you using ABS to build the kernel or building it isolated? On prebuilt Image.gz-dtb there are 20 dtbs found. Even when appending the stock dtbs, I'm not able to boot the kernel.

teknoraver, 2016-09-12:
I'm not using ABS, and it doesn't boots for me

chhapil, 2016-10-17:
i didnt get Image.gz-dtb when i unpacked the boot.img 
anybody has the luck?

chhapil, 2016-10-23:
any luck with bootable kernel?

chhapil, 2016-10-26:
@bgcngm did u manage to boot the kernel? I saw your CM kernel, I;m struggling to boot for stock