MiCode/Xiaomi_Kernel_OpenSource #76: Please release the Xiaomi Mi Pad 2 kernel sources!!!

LIVEDTEN, 2016-09-26:
Please release the Xiaomi Mi Pad 2 kernel sources!!!

ttkea, 2016-10-05:
是的，请开源代码！

endrix, 2016-11-14:
You are obliged by GPL to release the sources, what you are doing is illegal and you should be in court for that. This is totally unacceptable and Xiaomi should be ashamed for that, companies like yours destroys the open source.

DeckardShaw, 2016-12-08:
Mi pad sources-
 https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/latte-l-oss?files=1

tank0412, 2017-01-17:
This kernel sources is broken.
So it is useless

LIVEDTEN, 2017-01-17:
I confirm assembly crash, while the alternative does not work 100%

eturk1, 2017-01-17:
is anyone thinking a Chinese company is actually subject to any international agreement like GPL, etc? LOL Hard enough to get a S. Korean company to provide this