MiCode/Xiaomi_Kernel_OpenSource #80: Bootable Mi5 Kernel

chhapil, 2016-10-26:
Does anybody managed to create a bootable kernel from Xiaomi kernel source for Mi5?
Though I managed to compile it, cant boot it on device.
Tried with Global Stable and beta ROMs

Any help here appreciated

33bca, 2016-10-27:
Read on XDA, this kernel is only bootable with Miui 7 and its firmware. Miui blocks this kernel in the bootloader.

chhapil, 2016-10-28:
@Shaky156 managed to boot it on MIUI8 too. Just looking for pointers what could be the possible ways.

anarkia1976, 2016-11-01:
the problem seems related to kernel source and not to blocked bootloader ... but i think olny @Shaky156 can be point us to the possibile solution.

chhapil, 2016-11-02:
@Shaky156 has a bootable kernel built from caf source, not modified Xiaomi

rushabh71, 2016-11-02:
okay mine is not booting to miui 8 but in miui 7 it is booting what needs to be changed in source to make it boot in miui 8...

psndna88, 2016-11-02:
use different caf base for miui 8 MM

rushabh71, 2016-11-02:
@psdna88 what is caf? BTW i am new to this..:)

psndna88, 2016-11-02:
code aurora foundation sources for msm on which xiaomi adds device specific
files to get a working kernel..

16rd, 2016-11-02:
HI all,

     do you know which light sensor used in MI Max?  thanks.

liujin@new-mobi.com
 
From: Parvinder Singh
Date: 2016-11-02 19:25
To: MiCode/Xiaomi_Kernel_OpenSource
Subject: Re: [MiCode/Xiaomi_Kernel_OpenSource] Bootable Mi5 Kernel (#80)
code aurora foundation sources for msm on which xiaomi adds device specific
files to get a working kernel..

psndna88, 2016-11-02:
pull out config file from stock boot.img and u can see in sensors section which driver is used..

16rd, 2016-11-02:
Dear Parvinder Singh,

   Thank you, I wonder if I can see the in the source? 

liujin@new-mobi.com
 
From: Parvinder Singh
Date: 2016-11-02 19:37
To: MiCode/Xiaomi_Kernel_OpenSource
CC: 16rd; Comment
Subject: Re: [MiCode/Xiaomi_Kernel_OpenSource] Bootable Mi5 Kernel (#80)
pull out config file from stock boot.img and u can see in sensors section which driver is used..

anarkia1976, 2016-11-02:
Source online with commit history, error fixed:
https://github.com/anarkia1976/android_kernel_xiaomi_msm8996/commits/master

boot is ok but phone restart after bootanimation ...

psndna88, 2016-11-02:
u have u use newer caf base,  the one used in LP kernel. will not work for
MM
i have my working cm 14.1,  miui mm kernels for Redmi note 3 kenzo, on xda,
using caf newer sources and some device specific drivers from official Lp
kernel source given by xiaomi...

rushabh71, 2016-11-02:
@psndna88 sorry for disturbing again....
but this is what you are talking about...
https://wiki.codeaurora.org/xwiki/bin/QAEP/

after following above guide i have to make kernel from cm source code root directory.....
Thank you...

psndna88, 2016-11-02:
what linux kernel version is there for your device?  3.10.y right.?  

https://source.codeaurora.org/quic/la/kernel/msm-3.10/?h=msm-3.10

bgcngm, 2016-11-02:
No, Linux kernel 3.18.y.

https://source.codeaurora.org/quic/la/kernel/msm-3.18/?h=msm-3.18

psndna88, 2016-11-02:
ok u have to start with a branch from it

rushabh71, 2016-11-02:
@psndna88 THANK YOU SO MUCH IT REALLY HELPED

anarkia1976, 2016-11-02:
Source fixed:
https://github.com/anarkia1976/android_kernel_xiaomi_msm8996/commits/ak

chhapil, 2016-11-03:
what is the logic behind the tags?
https://wiki.codeaurora.org/xwiki/bin/QAEP/release
is higher version number is always the with the latest commits?

psndna88, 2016-11-03:
depending on chipset,  highest number is needed for latest Android
versions..

chhapil, 2016-11-04:
@psndna88 what all Xiaomi specific drivers are normally needed to be merged?

psndna88, 2016-11-04:
check out

https://github.com/psndna88/AGNi_pureMIUI/commits/AGNi_MM_kenzo?after=rVkJLlye%2Fa4C1RQPgKb42R%2Bf93QrMjc5

and previous page..

chhapil, 2016-11-04:
I'll check, thanks @psndna88

anarkia1976, 2016-11-04:
@psndna88 
your source is for redmi note 3, can you port your effort also for MI5?
thanks

psndna88, 2016-11-04:
LA.HB.1.1.2_rb1.9 branch  should work for you...  till 19th im not free..

chhapil, 2016-11-06:
No luck making bootable kernel

anarkia1976, 2016-11-06:
http://forum.xda-developers.com/showpost.php?p=69472067&postcount=101

anarkia1976, 2016-11-12:
Please source .... we need an official one! .....

anarkia1976, 2016-11-13:
https://www.change.org/p/xiaomi-release-updated-kernel-source-for-xiaomi-mi-5?recruiter=9804892&utm_source=share_petition&utm_medium=twitter&utm_campaign=share_twitter_responsive

Anik1199, 2016-11-13:
This won't help to be honest :3 at least if it is Xiaomi ;)
#RIPGPL

anarkia1976, 2016-12-02:
+1+2+3+4+5+6+7+8+9

nicoula, 2016-12-05:
#RIPGPL

anarkia1976, 2016-12-05:
We have to wrote directly to @xiaomi support. I don't think that look the open issue.
https://twitter.com/anarkia1976/status/805769517242679297

amitgrin, 2017-02-12:
can someone put a link to this Mi5 ROM with MIUI 7?
all the roms from the offical site was with MIUI8.

thanks.