MiCode/Xiaomi_Kernel_OpenSource #83: Source for Mi MIX Please!

Nokius, 2016-11-12:
Having sources for this device would be really nice!

Thanks in advance

infeedel, 2016-12-07:
Upvoted !

anetar, 2016-12-15:
Hopefully source version as soon as the suspended updated with other Source  versions I see oneplus company going in the way of adsorbed all its devices is  updated source
very soon Version to kernel sources and camera blobs

https://github.com/OnePlusOSS

I hope Xiaomi to become like oneplus

sh0w0ffin, 2017-04-27:
Now you can close issue, source released ( scorpio-m-oss ).

infeedel, 2017-04-27:
These are still scorpio sources and not lithium sources

luca020400, 2017-04-27:
@infeedel this release support every msm8996 device by xiaomi, including lithium

infeedel, 2017-04-27:
@luca020400 my bad, wasn't entirely up to date

Nokius, 2017-04-27:
request done with [commit  671c431 ](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/commit/671c431f6dd350fb702d2b0b1441f80af6af2903)

Thanks

closed 2017-04-27 by Nokius