MiCode/Xiaomi_Kernel_OpenSource #84: Source for MI5s

Gabriel-Campos, 2016-11-16:
Please Xiaomi, release the mi5s source asap!

nicoula, 2016-11-17:
+1.

Please let us know if you're going to do it or not at least.

wspl, 2016-11-18:
and Mi 5s Plus too. ☹

nazunamoe, 2016-11-21:
SOURCE PLEASE!

gtebbutt, 2016-11-21:
Would be great to get the source for the Mi5s

CL0SeY, 2016-11-25:
GPL request for kernel source for Mi5s please.

Thanks!

SuperDian, 2016-12-01:
Source Please.

anarkia1976, 2016-12-02:
+1+2+3+4+5

AdeonBlack, 2016-12-05:
Please release the kernel sources for the Mi 5s ASAP, you're not respecting the GPL

tcraxs, 2016-12-08:
capricorn source please.

anetar, 2016-12-15:
Hopefully source version as soon as the suspended updated with other Source  versions I see oneplus company going in the way of adsorbed all its devices is  updated source
very soon Version to kernel sources and camera blobs

https://github.com/OnePlusOSS

I hope Xiaomi to become like oneplus

keks2293, 2017-01-28:
+

emman-bautista, 2017-02-14:
It's been 4 months already after this is released and still we don't have the source of the kernel. I wonder how long this will take to release. They should respect the GPL requirements.

fadeeveu, 2017-02-14:
I join the multiple requests for the mi5s kernel. I hope xiaomi follow the GPL requirement stricter.

HITGIF, 2017-04-10:
release asap please +1

pkhodaveissi, 2017-04-13:
Please do release the source, It only hurts your popularity and make people think that they need to run away from your stock rom, at least this way people can enjoy your smart phones while you get a good enough stock rom and people would actually consider coming back to you

jant90, 2017-04-15:
+1 release please

anarkia1976, 2017-04-22:
+30

luk1337, 2017-04-22:
-29, no cheating pls.

anarkia1976, 2017-04-22:
I think we have to wait new mi6 or the new year :)

emman-bautista, 2017-04-22:
Dont count on xiaomi. They have the worst support ever for updates and
kernel release. I've arleady returned back to oneplus3t. Still the best
choice i made after xiaomi.

sh0w0ffin, 2017-04-27:
Now you can close issue, source released ( scorpio-m-oss ).

closed 2017-04-27 by Gabriel-Campos