MiCode/Xiaomi_Kernel_OpenSource #88: Broken MIPAD 2 (latte) defconfig

tank0412, 2016-12-21:
Hello.
It looks like you provided a broken defconfig in latte kernel sources because it is always restarting.
Can you re-check it?
Thanks.

harrynowl, 2016-12-25:
It doesn't seem to be the right defconfig at all. It compiles a bunch of modules we don't need, misses some we do and some hardware (i.e. wlan) isn't even configured.

tank0412, 2017-01-17:
> It doesn't seem to be the right defconfig at all. It compiles a bunch of modules we don't need, misses some we do and some hardware (i.e. wlan) isn't even configured.

It happened because defconfig is restarted. I mean in your case compiler even did not use defconfig by Xiaomi.
In your case defconfig is restarted and you got a generic defconfig for x86_64.
To check my statement check .config,

matyasf, 2017-01-31:
any update on this? Xiaomi why are you releasing code that does not even build? This is the quality you want to be know of?

harrynowl, 2017-01-31:
Yes, see my repository or the one on late-dev org.

matyasf, 2017-01-31:
Looks great!

I would like to cotribute, but I dont know how to begin.     
First how do you compile a ROM from those 3 repos? Are there some manuals/tutorials somewhere?   
Also as I have read the biggest issue is sound now. Cant you take the driver source/binaries from this repo to get it work?    
Also I've read somewhere that you are trying now to get a full 5.x build working first? Or are you only working on a 6.x ROM?

harrynowl, 2017-02-01:
Well I believe Roman merged the tree over at latte_dev to be unified now so you will need the kernel, device & vendor blob repos. Add each to your .repo/local_manifests/roomservice.xml. Then you can use brunch latte.

Working on kernel side sound fix as I don't think it loads firmware property in Xiaomi's provided source, but I don't get lots of time to work alongside university study.

eturk1, 2017-02-01:
ah, I see, you're referring to https://github.com/latte-dev
latte-dev here 
cool!

matyasf, 2017-02-05:
@harrynowl Thanks. So I checked out the repos and have ubuntu with stuff thats needed installed (repo, JDK 7, Git,...)  
Whats else needs to be in roomservice.xml? Can you please post yours? When I edit roomservice.xml and run "brunch latte" it will build the full ROM, I just need to flash it? Or is there something else left? 
This is my roomservice.xml now, I have no idea wheter it looks good:   

    <?xml version="1.0" encoding="UTF-8"?> 
    <manifest>
      <project name="latte/android_device_xiaomi_latte" path="android_device_xiaomi_latte"  />
      <project name="latte/android_kernel_xiaomi_latte" path="android_kernel_xiaomi_latte"  />
      <project name="latte/android_vendor_xiaomi_latte" path="android_vendor_xiaomi_latte"  />
    </manifest>

Also I have initialized repo with repo init -u https://android.googlesource.com/platform/manifest -b android-6.0.1_r78 is this correct?

Thanks!

tank0412, 2017-02-06:
Also you should manually make AOSP bring up in android_device_xiaomi_latte

matyasf, 2017-02-06:
What do you mean by "manually make AOSP bring up in android_device_xiaomi_latte"? after I ran brunch latte I need to run make in the android_device_xiaomi_latte folder?

BTW the repo init command above is not good since AOSP doesnt seem to have the brunch command. as I see its only in Cyanogenmod so now Im calling repo init -u https://github.com/CyanogenMod/android.git -b cm-13.0

harrynowl, 2017-02-06:
LineageOS (and similar) and AOSP have slightly differing makefiles, so you will have to adapt. See a google device bringup (e.g. Pixel) for reference.

harrynowl, 2017-02-06:
Also after running repo init you must repo sync to pull the source code to your computer.

tank0412, 2017-02-06:
Guys, i wrote a letter to a Xiaomi developer who published latte kernel sources in this Github repo.
And i got this answer:

> We will be testing after release latte kernel code, and it’s can be build successful, device can be working well after flash build boot.img.

So he answered that sources is not broken.

matyasf, 2017-02-09:
hmm then why cant we compile it? Do they use some special toolchain?

tank0412, 2017-02-10:
Toolchain can not affect defconfig.
It can affect building only.
So something really wrong with latte kernel sources

harrynowl, 2017-02-11:
I do wonder if it's worth rebasing the kernel to one of the releases for the CHUWI tablets from JideTechnology.

tank0412, 2017-02-11:
I think that it is needed because this way we will get a great base for next upstream updates.

matyasf, 2017-02-11:
I've googled a bit for what other device uses the same CPU and found this one: http://support.lenovo.com/hu/en/products/Tablets/Yoga-Series/YOGA-TAB-3-Pro-10?tabName=Downloads&linkTrack=Mast:SubNav:Support:Drivers%20and%20Software|Drivers%20and%20Software&beta=false The yoga 10 pro has the same CPU (z8500), but sadly just Android 5.1 was released into it (Lenovo promised 6.0, but fails to deliver)

harrynowl, 2017-02-11:
I'll look at rebasing it. @matyasf you generally only need same CPU series, cherry trail can be found in a few tablets- Konstat has done a fair bit of work for them. I think part of the 6.0 issue is lack of support from Intel.

tank0412, 2017-02-12:
Guys, i built RR MM build which is based on native latte MIUI LP blobs + ramdisk with my Marshmallow bring up.
Can somebody test it? Thanks.
https://drive.google.com/open?id=0B-Fin8UxrD6PUFFBRGJmeGVveU0

matyasf, 2017-02-12:
It does not work for me, it hangs endlessly on the RR boot animation. Heres an except from my kernel log:

[    4.998449] init: init second stage started!
[    4.998742] init: waitpid failed: No child processes
[    4.998880] init: (Loading properties from /default.prop took 0.00s.)
[    4.999768] init: (Parsing /init.environ.rc took 0.00s.)
[    4.999925] init: (Parsing /init.usb.rc took 0.00s.)
[    5.000754] init: /init.latte.rc: 824: ignored duplicate definition of service 'smonsir'
[    5.000796] init: /init.latte.rc: 830: ignored duplicate definition of service 'smprosrv'
[    5.000874] init: could not import file '/init.debug-charging.rc' from '/init.latte.rc'
[    5.000921] init: could not import file '/init.logs.rc' from '/init.latte.rc'
[    5.000971] init: could not import file '/init.crashlogd.rc' from '/init.latte.rc'
[    5.001022] init: could not import file '/init.coredump.rc' from '/init.latte.rc'
[    5.001077] init: could not import file '/init.diag.rc' from '/init.latte.rc'
[    5.032499] snd_soc_register_card failed -517
[    5.032507] platform cht_rt5659: Driver cht_rt5659 requests probe deferral
[    5.032651] ALSA device list:
[    5.032679]   No soundcards found.
[    5.744742] init: write_file: Unable to open '/proc/cpu/alignment': No such file or directory
[    5.744847] init: write_file: Unable to open '/proc/sys/kernel/sched_compat_yield': No such file or directory
[    5.745323] init: write_file: Unable to write to '/dev/cpuctl/cpu.shares': Invalid argument
[    5.745388] init: write_file: Unable to open '/dev/cpuctl/cpu.rt_runtime_us': Permission denied
[    5.745435] init: write_file: Unable to open '/dev/cpuctl/cpu.rt_period_us': Permission denied
[    5.745559] init: write_file: Unable to open '/dev/cpuctl/bg_non_interactive/cpu.rt_runtime_us': Permission denied
[    5.745606] init: write_file: Unable to open '/dev/cpuctl/bg_non_interactive/cpu.rt_period_us': Permission denied
[    5.747014] init: write_file: Unable to open '/proc/sys/abi/swp': No such file or directory
[    5.747565] init: /dev/hw_random not found
[    6.227540] init: cannot find '/system/bin/bd_prov', disabling 'bd_prov'
[    6.240463] cgroup: new mount options do not match the existing superblock, will be ignored
[    6.242825] init: property_set("ro.hwui.texture_cache_size", "24.0f") failed
[    6.242869] init: property_set("ro.hwui.text_large_cache_width", "2048") failed
[    6.242918] init: property_set("ro.hwui.text_large_cache_height", "512") failed
[    6.242969] init: property_set("ro.opengles.version", "196609") failed
[    6.243058] init: property_set("ro.sf.lcd_density", "320") failed
[    6.243131] init: property_set("ro.btconfig.if", "uart") failed
[    6.243167] init: property_set("ro.btconfig.dev", "/dev/ttyHSU0") failed
[    6.243218] init: property_set("ro.btconfig.vendor", "broadcom") failed
[    6.243282] init: property_set("ro.btconfig.chip", "BCM4356") failed
[    6.243321] init: property_set("ro.wlan.mimo", "1") failed

[   11.123628] init: cannot execve('/system/bin/otpserver'): No such file or directory
[   11.123728] init: Starting service 'sensorhubd'...
[   11.124247] init: cannot execve('/system/bin/keymaster_meid'): No such file or directory
[   16.149547] init: cannot execve('/system/bin/sensorhubd'): No such file or directory
[  297.181572] init: cannot execve('/system/vendor/bin/power_hal_helper'): No such file or directory
[  297.182114] init: Service 'power_hal_helper' (pid 5880) exited with status 127

(the init messages repeat endlessly)
Also I got the same message about the

tank0412, 2017-02-12:
@matyasf 
What about logcat?
And PM me via Hangouts. (My gmail for it you can find in my Github profile)

matyasf, 2017-02-12:
Heres the full logcat:
https://dl.dropboxusercontent.com/u/9186429/latte_logcat_02_12.txt

I've sent you an invite in Hangouts.

harrynowl, 2017-02-15:
Rebase work started. Hopefully it proves useful.

eturk1, 2017-02-15:
noticed this team trying to build the kernel also. maybe grow the team?
http://en.miui.com/thread-411400-1-1.html

tank0412, 2017-02-16:
@eturk1 It is not a team. And it looks like he stop do something for latte.

@harrynowl I found that audio firmware on latte blobs requers symlinks.
And after that audio logcat changed a lot. But audio logcat is different on your kernel and prebuilt by Xiaomi.
On your kernel something like this:
W RouteManager: audio-parameter-manager:         Unable to back synchronize configurable element /Audio/REALTEK/SOUND_CARD/INPUT/ANALOG/4/VOLUME: ALSA: Control element count (1) and configurable scalar element count (2) mismatch

W RouteManager: audio-parameter-manager:         Unable to back synchronize configurable element /Audio/REALTEK/SOUND_CARD/OUTPUT/LINEOUT/LOUT/RIGHT/MIX/DACR2/MONO_SWITCH: Failed to open mixer control: LOUT MIX DAC R2 Switch

Stock xiaomi kernel does not have this errors.
It have something liek this:
W/RouteManager( 2397): audio-parameter-manager: Unable to forward synchronize configurable element /Audio/REALTEK/SOUND_CARD/MIXER/DIGITAL/DAC_MIXER/STEREO/LEFT/L1/SWITCH: Card cherrytrailaud not found. Error: No such file or directory
D/TinyAlsaAudioDevice( 2397): open: card (cherrytrailaud, 0) 
D/TinyAlsaAudioDevice( 2397): 	 config (rate=48000 format=0 channels= 2).
D/TinyAlsaAudioDevice( 2397): 	 RingBuffer config: periodSize=1152 nbPeriod=4startTh=1151 stop Th=4608 silence Th=0
E/AudioUtils( 2397): Sound card cherrytrailaud does not exist
E/TinyAlsaAudioDevice( 2397): open: Cannot open tinyalsa (cherrytrailaud,0) device for output stream (error=cannot open device '/dev/snd/pcmC4294967294D0p': No such file or directory)
D/TinyAlsaAudioDevice( 2397): close
E/RouteManager( 2397): 	 error while routing Media=

So far and so near for audio fix. 

P.S. BTW, if someone wants to test few new experimental rom build, pm via Hangouts. (email you can find in my github profile)

harrynowl, 2017-02-16:
@eturk1 I noticed this but see no source code link for their device tree or kernel tree. Only the ones they forked from.

ATM I am away and can't use my device for testing things. So I'll just work on a rebase as it still makes Xiaomi's kernel nicer by committing changes properly.

eturk1, 2017-02-16:
> @eturk1 I noticed this but see no source code link for their device tree or kernel tree. Only the ones they forked from.

are they reachable?

tank0412, 2017-02-19:
Guys,  I FIXED AUDIO ON RR 5.7.4!

eturk1, 2017-02-20:
congrats @tank0412 on the audio fix.

harrynowl, 2017-02-22:
Nice, good effort :+1: 

I am home again so I can continue work..

tank0412, 2017-02-24:
Well, guys, i talked with KonstaT about dead sleep issue and he said that we got it due to Lollipop kernel. And he did a 

> completely new kernel bring up based on Intel Marshmallow BSP for CM13
So his chuwi tablet has no issues due to it.
It looks like we must make kernel rebase and add all latte drivers to there:
https://github.com/CM-CHT/android_kernel_intel_cherrytrail/tree/cm-13.0
The problem is we still do not have a working defconfig. 
It is sad.
It will be very hard.

So @harrynowl if you continue do a kernel rebase, pls use this source instead of sources from JideTechnology/remixos-kernel
https://github.com/CM-CHT/android_kernel_intel_cherrytrail/tree/cm-13.0

I tried to fix at least audio in current defconfig but i failed.

eturk1, 2017-02-24:
million thanks @tank0412 for hunting down the "dead sleep" bug! (and now a common name for it)

maybe by year end xiaomi has miui 9 for mi pad 2, we get sources, and unicorns return to earth. LOL
you may crack this before then!