MiCode/Xiaomi_Kernel_OpenSource #90: Release source for redmi 4 prime

anomalchik, 2016-12-27:
Please, release kernel source code for redmi 4 prime(snapdragon 625)

Anik1199, 2017-01-23:
Yeah please :)

shlyakpavel, 2017-01-26:
We had to wait redmi 3s kernel source for ages. I hope xiaomi will not make as wait for so long this time :(

TingyiChen, 2017-01-27:
+1.Need this to make stable LineageOS

FloSchramm, 2017-02-03:
Yes please release it soon! Thank you!

msdx321, 2017-02-04:
Since the source should be quite similar to land's one, please release it ASAP

alwinrosid, 2017-02-12:
wait

ArDrift, 2017-02-12:
Yes please, we need it!

IvanKozinov, 2017-02-12:
Awaiting too, thanks.

Anidetrix, 2017-02-19:
Agreed, release kernel source please!

baicaiC, 2017-03-03:
Please.

WYFXF, 2017-03-03:
please

jummmy, 2017-03-06:
please

orient33, 2017-03-06:
这么久了还不开源是为了什么，还是因为什么

gw1021, 2017-03-07:
please... 1+

crofyz, 2017-03-07:
please...+1

boyshawn, 2017-03-13:
I will definitely appeal to opening the source code for Redmi 4 Prime. We need it to develop our custom rom. And it is by GPLv3 licences that this must be released.

Pablohn26, 2017-03-14:
Yes please +1 . Redmi 4 prime aka Markw

luk1337, 2017-03-14:
tip: Leaving `+1` here will do /literally nothing/.

Pablohn26, 2017-03-14:
@luk1337 what would you suggest to do /literally something/ ?

luk1337, 2017-03-14:
Maybe email people responsible for OSS xiaomi stuff instead of doing stuff like: `+1`, change.org, etc.

Pablohn26, 2017-03-14:
@luk1337 there are several repos (this one for example is outdated). Do you have more information about that people?

luk1337, 2017-03-14:
Check who released source for branch X and get this guy's email address.

boyshawn, 2017-03-14:
@luk1337 The latest reply is [commit ee3f64a442955d5a19f5625e18cd6c98e655a853](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/commit/ee3f64a442955d5a19f5625e18cd6c98e655a853), and it is by @hehao <hehao@xiaomi.com>. So you mean we should email him?

luk1337, 2017-03-15:
Definitely not "we". One email is enough.

WYFXF, 2017-03-16:
I just want android n

rocketime, 2017-03-16:
asap plase

ArDrift, 2017-03-16:
@boyshawn I sent an email to hehao roughly a month ago, but there hasn't been any response since then.

nemorosus, 2017-03-17:
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/markw-m-oss

Is this it, or?

Anik1199, 2017-03-17:
Yes it is.