MiCode/Xiaomi_Kernel_OpenSource #411: release kernel source for mi5s plus

anetar, 2017-01-02:
please release kernel source for mi5s plus

paranoiagu, 2017-01-04:
please release kernel source for mi5s plus

anetar, 2017-01-20:
Hi, the N kernel source code is live at: https://github.com/OnePlusOSS/androi...eplus/QC8996_N

Grief, 2017-02-24:
Cannot @torvalds help us?

sh0w0ffin, 2017-04-27:
Now you can close issue, source released ( scorpio-m-oss ).

Grief, 2017-04-27:
@sh0w0ffin are you high? scorpio is mi note 2, not mi5s plus!

luk1337, 2017-04-27:
@Grief Are you high? look at the commit message, it's common kernel for all 8996 devices :^)

Grief, 2017-04-27:
@luk1337 @sh0w0ffin yes, I am. Didn't see the actual commit message 'Kernel: Xiaomi kernel changes for Mi 5/Mi Note2/Mi 5s/Mi 5s plus/Mi Mix '. Sorry about that. Congratulations then!

dksslq, 2017-05-14:
Can "scorpio-m-oss" compile and repackage it into boot.img and then flash it into Xiaomi 5s plus?,I am new.....

luk1337, 2017-05-14:
@dksslq no one stops you from doing that :3