MiCode/Xiaomi_Kernel_OpenSource #412: [kenzo]Planning to sue xiaomi

LightTube, 2017-01-03:
I'm planning to sue xiaomi for violation of GNU GPLv2 license terms. As marshmallow 6.0.1 for HMNote3Pro is already out but the updated kernel sources are not out yet. 

I will wait a week for them to update and after that i will sue them. 

Get ready for the notice.

TechExhibeo, 2017-01-03:
Nice Warning 😂😂😂😂😂😂
gave me goosebumps 😫

eturk1, 2017-01-04:
related to this?: https://www.kernel.org/doc/pending/gplv2-howto.html
this is interesting!!

jcadduono, 2017-01-08:
10/10

if u send them a well-written letter they may xerox and mail the source code back to you

DroidThug, 2017-01-08:
LMAO lets see what happens now :laughing: :laughing: :rofl: :rofl: :rofl:
*grabs some popcorn*
## :P ##

Bitti09, 2017-01-08:
grabs some :popcorn:  
xiaomi  answer to custom rom devs: " For now we don't have any plan for Custom Rom support"
so good luck

TechExhibeo, 2017-01-08:
@Bitti09 hahahahah 😂😂😂😂

eturk1, 2017-01-08:
In 1s and 0s u can type in :)

Bitti09, 2017-01-12:
@LightTube  any news  about your plan to sue xiaomi? :laughing:

shack-32, 2017-01-19:
And please just release source for latest devices like Capricorn, natrium, etc..

Daviteusz, 2017-01-19:
@LightTube Good idea, sue Xiaomi for breach of the GPL 2.0 licence

eturk1, 2017-01-19:
sue them, where, in a communist chinese court? LOL
ya, probably not a reality  to sue them :(

tailot, 2017-01-20:
:rofl: :rofl: :rofl: :rofl: :rofl: :rofl: :rofl:

anetar, 2017-01-20:
Hi, the N kernel source code is live at: https://github.com/OnePlusOSS/androi...eplus/QC8996_N

MSF-Jarvis, 2017-01-20:
Damn @anetar you're drunk AF

dragonGR, 2017-01-20:
@anetar , you're high bro.. high af.

jcadduono, 2017-01-20:
cant blame him, it's hard to tell these GPL slacking chinese companies apart

dragonGR, 2017-01-20:
Asking xiaomi to release the sources is like asking your wife for sex. You never know.

nolanroell, 2017-01-21:
Don't criticize him, let's all get together and form a class-action. 😂

Bitti09, 2017-02-06:
Asking chinese companies to release the sources is like talking with a stone. You never get an answer :rofl:

Bitti09, 2017-10-13:
@LightTube any Progress? 😂😂😂

_Sent from my Xperia XZ (AOSP) using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

eturk1, 2017-10-13:
how do you sue a communist company in a communist court?
have you not noticed their mascot wearing the cute red-star communist hat?
[image]