MiCode/Xiaomi_Kernel_OpenSource #413: [nikel] Redmi Note 4 Kernel Sources

MimarSinanT, 2017-01-08:
Please release kernel sources for Redmi note 4.
Vernee Apollo lite mt6797 source code released but we can't use it because it's missing vendor config to build a working kernel for Redmi note 4, also missing a lot kernel driver for Redmi note 4 (LCM, Touchscreen, Camera, Fingerprint...).

anetar, 2017-01-20:
Hi, the N kernel source code is live at: https://github.com/OnePlusOSS/androi...eplus/QC8996_N

MimarSinanT, 2017-01-26:
@anetar,No relevance to what I want

krizky82, 2017-03-02:
I probably won't but any xiaomi device with mediatek processor ever again 😑
Please release the kernel source...

MimarSinanT, 2017-03-04:
Yes, this is so such.

Reisya999, 2017-03-22:
yes, please release kernel

jorgeiba97, 2017-03-30:
release it please

iulianvrz, 2017-04-02:
+1 we need this

Adilhusain, 2017-04-23:
Hi....
I am Adilhusain
I request you to release a kernal source of redmi note 4 (indian version)
Please Release It 
its heavily needed to build my own Rom

Supermatt01, 2017-05-18:
This device is from last year. What's the hold up? Release the source code in full already. Many people who would develop for your mtk devices are saying they'll never get an mtk version device from Xiaomi again.

jorgeiba97, 2017-06-19:
release it please

nedpals, 2017-07-05:
If u really respect GPLv2, you shouldve release the source already

_Sent from my Xiaomi Redmi Note 4 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

MimarSinanT, 2017-09-05:
I'm selling this damn phone and I'm going to buy a new SD phone. Do not wait for resources after this, as the phone is already out of date, the fashion of your phone will be over.

SORRY > TRANSLATE...

Reisya999, 2017-12-20:
saya juga sudah menjualnya, saya ganti dengan surge s1

raflyif, 2018-10-16:
Fuck

raflyif, 2019-03-07:
hello

seanlinmt, 2019-03-15:
release the kraken!!

barfin, 2019-11-15:
i guess there is no hope for this device

EpicStep, 2019-12-18:
Relese kernel please.

ashtasht, 2020-05-21:
Release the kernel please

lineagefan, 2020-09-06:
You guys can still do this, please release the kernel sources. It's still an decent processor for a phone.

Waxaranai, 2020-12-26:
r.i.p nikel

Supermatt01, 2020-12-27:
I am not getting another MTK device if I can help it.