MiCode/Xiaomi_Kernel_OpenSource #417: Source for Mi 5S

MrCskncn, 2017-01-27:
Its been months! Need the source.

paranoiagu, 2017-02-03:
Also need the source for Mi5S Plus

tcraxs, 2017-02-03:
This is a GPL request of the kernel sources for Mi5S, please be GPL conform and release sources.

magetron, 2017-02-04:
Yes Need the source!

emman-bautista, 2017-02-14:
maybe GPL committee doesn't care then. Coz they cannot push Xiaomi to release kernels.

tjonsson, 2017-02-15:
Yes! Need the source!

sveip, 2017-02-24:
Is anyone from Xiaomi on this github report page at all?

sveip, 2017-02-24:
Last commit is from 2014... So I guess this is abandoned...

Grief, 2017-02-25:
@sveip I thought the same but you are looking into master branch which wasn't updated for a while. gemini (mi5) branch was updated 6 Sep 2016.

rashyyd, 2017-03-20:
I hopeful they'll release soon

paranoiagu, 2017-03-27:
when release the kernel source？

sh0w0ffin, 2017-04-27:
Now you can close issue, source released ( scorpio-m-oss ).

Grief, 2017-04-27:
@sh0w0ffin are you high? scorpio is mi note 2, not mi5s!

rashyyd, 2017-04-27:
It's there... But merge together with Mi Mix, and many other

sveip, 2017-04-27:
@rashyyd are you sure? If so, good news. Should have 3rd party andriod flavor builds soon then.

luk1337, 2017-04-27:
Uh, if it was just about the kernel source then you'd have them already. I have been running my own Mi 5s Plus builds for over a month without a kernel source lol.

Grief, 2017-04-27:
@luk1337 did you get all the hardware (camera, fingerprint sensor, sound etc.) working?

luk1337, 2017-04-27:
Yes.

Grief, 2017-04-27:
@luk1337 why didn't you share your work on XDA then? the best lineageOS rom ATM still has issues with hw.

luk1337, 2017-04-27:
I did.

luk1337, 2017-04-27:
Most of issues are related to lineage specific kernel / OS customizations. The hardware itself works just fine.

Grief, 2017-04-27:
@luk1337 oh you are THAT guy who did it. I didn't track your progress for some time and I am surprised that you've already fixed almost all the previous issues without the kernel sources. Great job!

Personally I was waiting for kernel sources not so much only to be able to install custom rom (i.e. lineage) but to be able to apply sultanxda patch: https://github.com/sultanxda/android_kernel_oneplus_msm8996/commit/abc05b16bbd33521c2fffaf491c5657a94bfcfc5
Not sure it still works though

luk1337, 2017-04-27:
You can hack the system/core/init for that thing. I did that once just to check something.

Grief, 2017-04-27:
@luk1337 I am not a low-level developer so I'd appreciate if you share an instruction how to bypass safetynet check for unlocked bootloader without patching the kernel

luk1337, 2017-04-27:
1. Find where init loads androidboot properties
2. Add a strcmp guard for verifiedbootstate
3. Rebuild boot.img
4. Profit?

rashyyd, 2017-04-27:
http://en.miui.com/thread-594646-1-1.html  there is the news we got

dksslq, 2017-04-27:
com'on

sveip, 2017-04-28:
@luk1337, I'm not sure if I understand what you mean. But getting proper kernel source is nothing but good news and will for sure make 3rd party ROM dev. much better and stable compared to using pre-built kernel.

luca020400, 2017-04-28:
@sveip more stable, unlikely
Open up the potential, with /shitty/ custom kernels sure ...
It just allow us to properly have a kernel that could be adapted to the upcoming Android versions ( for example we already rebased this tree on top of latest 7.1 caf release )

Grief, 2017-04-29:
@luk1337 I finally got some time to dig into what you've advised to me. I tried to find some more information in the net, but didn't find even a mention of the solution you've provided, let alone a detailed tutorial. But I think I get your idea. At least I tried to. Anyway, I think the second step of your really short explanation is to replace line 642 with 

    } else if (!strncmp(name, "androidboot.", 12) && strncmp(name, "androidboot.verifiedbootstate", 12) && name_len > 12) {

in https://android.googlesource.com/platform/system/core/+/android-4.2.1_r1.2/init/init.c

But as I said I am not a low-level developer and I'll really appreciate if you tell me where can I get the source of MIUI's `init` (or, maybe it's safe to take AOSP ones?) and how to build `init` only. I guess I can grab boot.img from device, extract it, extract initrd.img from it, and replace init there, am I correct?

luk1337, 2017-04-29:
[image]
Here's a protip for ya.

Grief, 2017-04-29:
LOL. Did it just before you've posted that
[image]

Grief, 2017-04-29:
@luk1337 but could you please answer the actual question I've asked? How to build `init` only and is it safe to take AOSP sources for it?

luk1337, 2017-04-29:
BTW idk if you can do that in miui. I doubt that AOSP init would work.

Grief, 2017-04-29:
@luk1337 that was my original feel but I though that I misunderstand something. So your method is inapplicable to MIUI while with these kernel sources I can fix that in kernel, build it, repack and install, right?
>  was waiting for kernel sources ... to be able to apply sultanxda patch

luk1337, 2017-04-29:
I guess so, I didn't think you were talking about MIUI lol.

Grief, 2017-04-29:
>Personally I was waiting for kernel sources **not so much only to be able to install custom rom** (i.e. lineage) but to be able to apply sultanxda patch
@luk1337 sorry, English not my native.
BTW yesterday I tried your [ROM](https://forum.xda-developers.com/mi-5s-plus/development/rom-lineageos-14-1-t3574738) and found a couple of issues. Just if these are not known, first one is related to [alarm while device is off](https://github.com/LineageOS/android_device_xiaomi_natrium/issues/5) and the second is camera crash and no way to restart it except deleting its data. I didn't raise a ticket since it is so obvious so it must be either already known or non-producible - I just changed camera (frontal/back) and switched mode (photo/video) several times in a row.

luk1337, 2017-04-29:
Oh, I know about the camera crash. It basically happens when you try to select mono camera in the recording mode.

Grief, 2017-04-29:
Maybe! I didn't use lineageos since cyanogenmod 14 so icons were not too obvious to me - I just opened a camera, and started to click buttons one-by-one to get what they do - I remember I clicked to turn on frontal camera, then switched to video recording (IIRC it was started on back camera) and I clicked the button again. It might be not the same button as in photo mode but I think it was on the same place as switch cameras button was. It could be mono mode button.

Grief, 2017-04-29:
@luk1337 didn't you try to compile this kernel? I cannot get it working. I compiles, I repack it into boot.img and flash with TWRP but device stucks on Mi logo. If you know a way to troubleshoot such issues, please advise me

luk1337, 2017-04-29:
I didn't try to do that on miui.