MiCode/Xiaomi_Kernel_OpenSource #420: Release source for redmi 4 (prada)

EvgeniyBuchiy, 2017-02-19:
Please, release kernel source code for redmi 4 prada (Qualcomm Snapdragon 430)

jacoghi, 2017-04-14:
Yep, waiting for it as well.

Y-Bee, 2017-05-21:
Please release source code redmi 4 prada

Y-Bee, 2017-05-23:
Hello!

I have purchased phone Xiaomi Redmi 4 (prada)which is produced by your company and its firmware contains Linux, busybox etc, that is free software distributed under GNU General Public License. I would like to get the source code of the modified components of these programs, since the GPL requires you to release it. How I can do it?

luk1337, 2017-05-23:
No one cares about GPL here, see ya in 6 months.

Kizoky, 2017-06-23:
Too bad prada isn't identical to 3S

Kizoky, 2017-11-13:
Please release it

NoPH8, 2018-10-02:
> No one cares about GPL here, see ya in 6 months.

More than year passed from your comment but even for now there are no sources for prada.
Xiaomi has disgusting software support.

Kizoky, 2018-10-02:
You are better off buying a new phone, really. I won't buy any Xiaomi phone anymore