MiCode/Xiaomi_Kernel_OpenSource #647: Going to Consumer Complaints Court

thescream, 2017-04-05:
Its been like forever since Redmi Note 3 Kenzo/Kate got Stable Marshmallow, you ignorant bitches not updating kernel source! Cunts release it fast otherwise i have no option

luca020400, 2017-04-05:
God dammit stop it, spamming their git won't magically release the kernel

thescream, 2017-04-05:
@luca020400 I'm out now mate! Only god can convince Xiaomi

GuaiYiHu, 2017-04-06:
Stop! You have a bad quality.

luk1337, 2017-04-06:
@GuaiYiHu just leave this idiot alone.

Sourav-21, 2017-04-10:
@thescream WTF you have done!!!

luk1337, 2017-04-10:
@Sourav-21 other than spamming /issues? Nothing.
What will he do? Nothing as well.

DoveKI, 2017-04-18:
@thescream Xiaomi devices open source code will not be updated after the code, do you have to see other xiaomi devices update kernel source code?
So fuck Xiaomi!

DoveKI, 2017-04-18:
@GuaiYiHu Hi baby

poad42, 2017-05-15:
If you are in the US its possible GPL is an enforceable contract

https://www.xda-developers.com/us-district-court-rules-gnu-gpl-is-an-enforceable-contract/

GuaiYiHu, 2017-05-15:
@poad42 Xiaomi is a Chinese company.... So many people are waiting for those kernel that hasn't been released...