MiCode/Xiaomi_Kernel_OpenSource #648: Please Update Mi5 kernel source code

Sobuj53, 2017-04-18:
I humbly request the developer of xaiomi to update the Mi5 kernel source. Mi5 is an awesome device and with updated kernel source user can do much more with it.

GuaiYiHu, 2017-04-18:
So is xiaomi note lte.

lkthomas, 2017-05-05:
correct me if I am wrong, but Kernel isn't something you want to update everyday, it's basically the interface between software and hardware, if hardware don't change, why kernel need to change?

SuperWangCC, 2017-06-05:
@lkthomas but it's necessery to update if the sources isn't perfect now.And it must follow the latest Android feature.In my opinion the best example for phone company is Google Nexus5.Its kernel update continued for sever years!!!

closed 2019-08-24 by Sobuj53