MiCode/Xiaomi_Kernel_OpenSource #651: Please upload Xiaomi Redmi Note 2 ( hermes ) kernel source code

banonymous, 2017-04-25:
A big amount of Xiaomi Redmi Note 2 ( hermes ) users are waiting for kernel sources. 
We really need it. 

We have more than 400 users in Telegram chat that want Marshmallow and Nougat update for Redmi Note 2 (hermes):
[https://t.me/joinchat/AAAAAEAcYGMW9RpRIW2-XQ](url)

If you wont release it within 3 moths we will make a group lawsuit for violating GNU GPL.
Please, release kernel source.

luk1337, 2017-04-25:
>If you wont release it within 3 months we will make a group lawsuit for violating GNU GPL.

I kinda doubt you'll do it. In fact I am pretty sure you won't do it.

banonymous, 2017-04-25:
> I am pretty sure you won't do it.

Maybe I would not, but 420 angry hermes users from all around the world will cooperate to do it,

luk1337, 2017-04-25:
Haha, 420 angry teenagers vs Xiaomi. Good luck dude.

eturk1, 2017-04-25:
do people forget xiaomi is in a Chinese Communist country. You really think anyone outside China would be listened to? LOL

banonymous, 2017-04-25:
> Haha, 420 angry teenagers vs Xiaomi. Good luck dude.

че ты ржешь ебанат?
Are you working in Xiaomi? No? Then why you are such a sarcastic douchebag? 
Suggest us better solution. If no - get to fuck out of here.

> do people forget xiaomi is in a Chinese Communist country. You really think anyone outside China would be listened to? LOL

It is a fuck up.
I will never buy a xiaomi device.

luk1337, 2017-04-25:
I am just a dude who's bored enough to shit post in MiCode/issues heh. BTW I am waiting for a source for a device I am working on too~

GuaiYiHu, 2017-04-30:
@luk1337 I'm waiting for an update for the source for my device I'm working on as well. And XiaoMi never listens to Chinese too...

luk1337, 2017-04-30:
That's really sad. Fortunately source for my device has been released :^)

GuaiYiHu, 2017-04-30:
@luk1337 Dear guy, can you help me with my device which is called mi note lte(virgo).
This device shares the kernel source with xiaomi 3/4(cancro).
It was upgraded to Android 6.0 and xiaomi has made some changes on this device as well...
So the kernel source which was pushed two years ago was too old. 
The sensors don't work...
The drivers of sensor is too old so these files cannot be found with the old kernel.
https://github.com/xiaomi-note-dev/android_device_xiaomi_virgo/commit/de078dc1cc2b3a3daa6bc3b3d005305efe36e6ea
And seems another version of xiaomi note has similar files... 
This is how it works...
https://github.com/xiaomi-note-dev/android_kernel_xiaomi_leo/tree/cm-14.1/drivers/iio/imu/garnet
These can be transplanted to cancro's kernel source...
https://github.com/xiaomi-note-dev/android_kernel_xiaomi_virgo
 I need help...

GuaiYiHu, 2017-04-30:
@luk1337 Dear guy, can you help me with my device which is called mi note lte(virgo).
This device shares the kernel source with xiaomi 3/4(cancro).
It was upgraded to Android 6.0 and xiaomi has made some changes on this device as well...
So the kernel source which was pushed two years ago was too old.
The sensors don't work...
The drivers of sensor is too old so these files cannot be found with the old kernel.
xiaomi-note-dev/android_device_xiaomi_virgo@de078dc
And seems another version of xiaomi note has similar files...
This is how it works...
https://github.com/xiaomi-note-dev/android_kernel_xiaomi_leo/tree/cm-14.1/drivers/iio/imu/garnet
Can these be transplanted to cancro's kernel source?
https://github.com/xiaomi-note-dev/android_kernel_xiaomi_virgo
I need help...

luk1337, 2017-04-30:
Sorry, I don't work on phones I don't own.

GuaiYiHu, 2017-04-30:
So sad :(...
But can what I just said be achieved?
I hope you can take a while to see those...