MiCode/Xiaomi_Kernel_OpenSource #656: Cannot build kernel for mi5s plus

Grief, 2017-05-01:
It builds without errors, but after repacking it into `boot.img` with `abootimg` and flashing, device shows 'mi' logo for ~30 seconds and shuts down. `/proc/last_kmsg` is empty and I have no idea how to troubleshoot that. Any help would be appreciated.

luca020400, 2017-05-01:
Can you run ```lsusb``` and post the output?
Anyways I think we got this kernel booting on MIUI

Grief, 2017-05-01:
@luca020400 sorry, don't get your point, how is `lsusb` relevant?

luca020400, 2017-05-01:
To know if it's similar to a problem we had on LineageOS where the device was crash-booting to EDL

Grief, 2017-05-01:
You mean flash 'broken' `boot.img`, and provide output of `lspci` from pc when phone is connected via USB cable?

luca020400, 2017-05-01:
Exactly

luca020400, 2017-05-01:
Also what Image did you repack?

Grief, 2017-05-01:
I didn't repack image - I just reflashed `boot.img` for my current 7.4.6 global dev ROM.

Grief, 2017-05-01:
@luca020400 it appears as `Bus 001 Device 011: ID 18d1:d00d Google Inc.` and doesn't automatically reboot. Seems that it enters fastboot with the kernel built by me when the cable is plugged in... Can I grab some logs from fastboot mode?

Grief, 2017-05-01:
@luca020400 I am building in sterile dockerized environment, I can share the setup with you if you have a spare minute to look at it. Maybe I did some obvious mistake during the build? I am pretty new to this android low level stuff.

Grief, 2017-05-02:
@luca020400 here is what I was talking about, the dockerized build environment. I'd appreciate if you take a look into scripts. Maybe I am missing something: https://github.com/Grief/xiaomi-mi5s-plus

luca020400, 2017-05-02:
https://android.googlesource.com/platform/prebuilts/gcc/linux-x86/aarch64/aarch64-linux-android-4.9/

Grief, 2017-05-02:
@luca020400 OK, I try now. In the meanwhile, please notice that `aarch64-linux-androidkernel-*` are just symlinks to `aarch64-linux-android-*`, I doubt that it would help

Grief, 2017-05-02:
@luca020400 I tried to follow your advice. Here is the changes I did in my environment to user pre-built toolchain v4.9: https://github.com/Grief/xiaomi-mi5s-plus/compare/testing However, it didn't help - the result is the same: 'mi' logo for ~30 seconds and automatic shut down. When the device is connected via cable to PC it appears to be running in fastboot mode. In that case it doesn't automatically shut down and remains in fastboot mode forever.

Grief, 2017-05-06:
Since this repository was looking more like a from time to time dump of kernel sources rather than an active project, I didn't expect to find a README here. However I was wrong, there is one: https://github.com/MiCode/Xiaomi_Kernel_OpenSource/wiki/How-To-Use and it contains some steps like `Change KERNEL_DEFCONFIG` which I didn't do... I'll give it another try now!

Grief, 2017-05-07:
Nope, doesn't boot either. Spent ~3 hours only to download the whole Qualcomm Android Enablement Project(QAEP). It is a little bit more handy than a toolchain since it directly produces boot.img, but this boot.img causes the phone to hang on 'mi' logo as well. The logo slightly more bright though than with the previous try, not sure why.

Grief, 2017-05-07:
Oh! It actually boots! But not working now. At least I can finally look to logs. Seems bright display indicates end of kernel boot and pass to Zygote or something . However the logs are pretty different from the stock one. And full of that stuff:
```[  324.874006] type=1400 audit(1494119481.418:139): avc: denied { ioctl } for pid=567 comm="Binder:483_2" path="socket:[124388]" dev="sockfs" ino=124388 ioctlcmd=7704 scontext=u:r:surfaceflinger:s0 tcontext=u:r:surfaceflinger:s0 tclass=unix_stream_socket permissive=0
[  324.874080] type=1400 audit(1494119481.418:140): avc: denied { ioctl } for pid=567 comm="Binder:483_2" path="socket:[124388]" dev="sockfs" ino=124388 ioctlcmd=7704 scontext=u:r:surfaceflinger:s0 tcontext=u:r:surfaceflinger:s0 tclass=unix_stream_socket permissive=0
[  324.910896] init: Service 'zygote' (pid 22007) killed by signal 9
[  324.910915] init: Service 'zygote' (pid 22007) killing any children in process group
[  324.910959] init: write_file: Unable to open '/sys/android_power/request_state': No such file or directory
[  324.910988] init: write_file: Unable to write to '/sys/power/state': Invalid argument
```

I'll try to investigate. Thank god, I have clues now and can reach `dmesg`.

Grief, 2017-05-08:
Finally! I built kernel exactly as wiki page says, but then packed boot.img aith `abootimg` from built `kernel` and `ramdisk` taken from the device - it work now. However it still doesn't pass CTS profile check (safetynet), but this is another story

closed 2017-05-08 by Grief