MiCode/Xiaomi_Kernel_OpenSource #657: [land-m-oss] Update kernel source with latest MIUI release

danielstuart14, 2017-05-05:
Hi,
    Seems like land (Redmi 3s) source code is out of date. Latest global MIUI (8.2.4.0) doesn't boot with this source. (It actually starts the boot process, and then restart when it is almost done.)
    Also, the kernel.zimage size is different. I didn't check yet if there were changes in DTB, but, AFAIK all the changes should be released with their sources, following the GPL license.

So, please, update land source code to match latest MIUI global release.

darshan1205, 2017-05-05:
Better upgrade it to 89xx CAF tag

danielstuart14, 2017-05-05:
That would be nice, but I'm only asking them to sync It with their latest release. That's a legal obligation based on GPLv2 license. 

If xiaomi think that they're better than any other android-based manufacturer, then, they should do like Apple and build their OS based in another license.

luk1337, 2017-05-05:
You won't gain much by crying about GPL2 here. Good luck~

danielstuart14, 2017-05-05:
I already found the issue (it's cuz cpuset is enabled in newer miui releases).
We can see how they care about OSS, it's very sad that one of the biggest names in the Android world doesn't give a **** about It. 

It's funny how their CEO says about how they love their community.

GuaiYiHu, 2017-05-11:
Xiaomi just release a bootable vertion then it won't upgrade this any more............