MiCode/Xiaomi_Kernel_OpenSource #658: Please release Redmi 4A ( Rolex) kernel source

wasefali, 2017-05-07:
Its is the best selling device of Xiaomi in India as kf now.  And have a large no of users.  We want further developments on it. Its only possible if you release source. PLEASE RELEASE IT.. please

NieDzejkob, 2017-07-23:
It's not "please release it". It's "because the linux kernel is licensed under GPLv2, each released binary has to be accompanied by the corresponding source code". As you might or might not know, failing to do so can get you in court. [We know many cases of successful GPL enforcement](https://en.m.wikipedia.org/wiki/Gpl-violations.org).

wasefali, 2017-07-24:
thanks for pointing this , i knew it though but I was trying to be polite
if it helps but these chinese dont give a fuck , they should be treated
harshly