MiCode/Xiaomi_Kernel_OpenSource #665: Release Mido's kernel source

RaviKumawat, 2017-05-08:
Maa ke lodo, behen ke tako, madarchodo batao kernel source kab
release kar rhe ho.
Release the kernel source asap or provide an ETA of the release you
fucking lazy assholes and cunts. You and your fucking lazy team of
devs who does nothing except for writing changelogs, at
least you can release the kernel source so we can do something
by ourselves. Stop fucking your mother and do some work.

luk1337, 2017-05-08:
Get out.

RaviKumawat, 2017-05-08:
Fuck you

luk1337, 2017-05-08:
How nice :3

RaviKumawat, 2017-05-08:
🤗

luk1337, 2017-05-08:
Anyway, I hope they delay the release date just cause of this :^)

TheScarastic, 2017-05-08:
😂😂😂

RaviKumawat, 2017-05-08:
Om boom bhakshm swaha luk😂

luk1337, 2017-05-08:
Sorry, I don't speak this garbage. Please use English next time you want to say something rude to / about me.

luca020400, 2017-05-08:
The engrish is strong here.

MSF-Jarvis, 2017-05-08:
Not sure why this kind of shit is even attempted, knowing that this has absolutely no positive impact on the cause.

Anik1199, 2017-05-08:
So he just made his account to post this **_shit_** here.

akhilnarang, 2017-05-08:
Because this is totally gonna make them release the source any faster......

ab123321, 2017-05-08:
ANGERY

Warmachine28, 2017-05-08:
TF is the description of the issue... me ded .-.