MiCode/Xiaomi_Kernel_OpenSource #667: But my make failed

dksslq, 2017-05-15:
kernel/sysctl_binary.c:140:13: error: ‘KERN_BOOT_REASON’ undeclared here (not in a function)
  { CTL_INT, KERN_BOOT_REASON,  "boot_reason" },
             ^~~~~~~~~~~~~~~~
kernel/sysctl_binary.c:527:13: error: ‘NET_IPV6_ACCEPT_RA_PREFIX_ROUTE’ undeclared here (not in a function)
  { CTL_INT, NET_IPV6_ACCEPT_RA_PREFIX_ROUTE, "accept_ra_prefix_route" },
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Is that my problem of operation？？How can I compile successfully?

dksslq, 2017-05-15:
the error is in scorpio-m-o

GuaiYiHu, 2017-05-15:
which device

dksslq, 2017-05-15:
xiaomi 5s plus

harrynowl, 2017-05-15:
What toolchain?

dksslq, 2017-05-15:
this
http://releases.linaro.org/components/toolchain/binaries/6.3-2017.02/aarch64-linux-gnu/gcc-linaro-6.3.1-2017.02-x86_64_aarch64-linux-gnu.tar.xz

dksslq, 2017-05-15:
I also failed to compile the code on https://github.com/LineageOS/android_kernel_xiaomi_msm8996

There may be a problem with the source's configuration

dksslq, 2017-05-15:
how shuld I configuration it?

GuaiYiHu, 2017-05-16:
Good luck.

harrynowl, 2017-05-16:
Use the 4.9 toolchain from google, I bet  it works. Alternatively you probably need to look at something like: https://github.com/Gateworks/linux-imx6/commit/b2d3c9bd93962b8c7cf69387fbaadc2909ec1787

dksslq, 2017-05-16:
thank you~I am going to try again。

closed 2018-04-15 by dksslq