MiCode/Xiaomi_Kernel_OpenSource #670: Xiaomi Redmi 3S (European version)

DiDHack, 2017-05-30:
Hello! Can Somebody help me?

So, I built the kernel from the source. All was be fine before I try to use my kernel in european version of Redmi 3S. Modem force the system to restart. Log:
```
[ 80.159608] binder: 1394:1783 transaction failed 29189, size 448-0
[ 86.506974] Fatal error on the modem.
[ 86.506994] modem subsystem failure reason: lte_ml1_rfmgr_stm.c:6675:Assert rf_driver_stm_error failed: .
[ 86.507004] subsys-restart: subsystem_restart_dev(): Restart sequence requested for modem, restart_level = SYSTEM.
[ 86.507063] M-Notify: General: 8
```
Chinise version is working fine. But Xiaomi has successful built kernel that works fine in chinise and european versions. I think I don't switch on something in the menuconfig but what?

P.S. Where Can I find a full defconfig for Redmi 3S? Because defconfig in the source is only for SoC (without touchscreen, led, IR and etc.). And I configure it piece by piece.

GuaiYiHu, 2017-06-08:
See zwliew's work.

DiDHack, 2017-08-13:
I have tried it but nothing