MiCode/Xiaomi_Kernel_OpenSource #677: Dear Xiaomi, please release source code for redmi 4x [santoni]

TSRsunrise, 2017-07-07:


TSRsunrise, 2017-07-07:
Dear Xiaomi,please release source code for redmi 4x [santoni]

TingyiChen, 2017-07-10:
we can port one.now only touchscreen not working

Nikesh001, 2017-07-15:
Please released sources code ASAP

Adipatiarya, 2017-07-17:
Xiaomi mang laris di indonesia.  tp sangat di sayangkan terlalu lama rilis kernel untuk tipe model2 keluaran baru. kaya redmi4x ini. 👎

adrenochrome25, 2017-07-17:
It's very probable that they wait for the new version based on android 7.0 to go stable/in production else they'll have to update it again in a few weeks

luca020400, 2017-07-17:
@adrenochrome25 yea sure lol

luk1337, 2017-07-17:
I wonder if people posting in their own garbage language actually think this is a good idea before pressing the 'comment' button lol.

TSRsunrise, 2017-07-24:
люк, нахуй пошел, упездок ссаный

ohjames, 2017-08-19:
@luk1337 Bahasa is not a "garbage language", it's beautiful. But hey continue calling languages garbage if you wanna be seen as an immature racist.