MiCode/Xiaomi_Kernel_OpenSource #681: Source Kernel for Redmi 4A

CarlosArriagaCM, 2017-07-12:
I hope and soon release source for Redmi 4A (rolex)

noiver, 2017-07-15:
I hope and soon release source for Redmi 4A (rolex)

CarlosArriagaCM, 2017-07-18:
same :v here

NieDzejkob, 2017-07-23:
I'll repeat what I said on an older issue (slightly expanded):

It's not "please release it". It's "because the linux kernel is licensed under GPLv2, each released binary has to be accompanied by the corresponding source code - you need to state each added, changed or removed line" (technically it has to be available as soon as the binary,  but you can't fix the past). As you might or might not know, not providing the source code can get you in court. [We know many cases of successful GPL enforcement](https://en.m.wikipedia.org/wiki/Gpl-violations.org), but hopefully Xiaomi will collaborate. This applies to all your devices, not only the 4A. You include the license text in the ROM, but it seems like you don't read it yourself. Please open your eyes before it's too late.

luk1337, 2017-07-23:
Wow, you're so serious about this gee pee el thingy. Too bad nothing will happen~

NieDzejkob, 2017-07-25:
we'll see

azzazza, 2017-08-30:
I'll repeat what I said on an older issue (slightly expanded):

It's not "please release it". It's "because the linux kernel is licensed under GPLv2, each released binary has to be accompanied by the corresponding source code - you need to state each added, changed or removed line" (technically it has to be available as soon as the binary, but you can't fix the past). As you might or might not know, not providing the source code can get you in court. We know many cases of successful GPL enforcement, but hopefully Xiaomi will collaborate. This applies to all your devices, not only the 4A. You include the license text in the ROM, but it seems like you don't read it yourself. Please open your eyes before it's too late.

kiodo1981, 2017-10-11:
I hope and soon release source for Redmi 4A (rolex)
https://www.change.org/p/xiaomi-inc-publish-the-kernel-source-for-the-xiaomi-redmi-4a-4-4pro