MiCode/Xiaomi_Kernel_OpenSource #686: Land kernel source is full of problems, or what happens here?

MPeti1, 2017-07-18:
As many times i download again the land kernel and try to compile, i have as many kind of problems...
The last time i got that, what is the problem?

`root@kali:~/Xiaomi_Kernel_OpenSource# make msm8937_defconfig
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
drivers/input/touchscreen/FT5336/Kconfig:2:warning: ignoring type redefinition of 'TOUCHSCREEN_FT5336' from 'boolean' to 'tristate'
drivers/input/touchscreen/IST30xx/Kconfig:2:warning: ignoring type redefinition of 'TOUCHSCREEN_IST30XX' from 'boolean' to 'tristate'
drivers/soc/qcom/Kconfig:329:warning: choice value used outside its choice group
drivers/soc/qcom/Kconfig:334:warning: choice value used outside its choice group
***
*** Can't find default configuration "arch/x86/configs/msm8937_defconfig"!
***
scripts/kconfig/Makefile:105: recipe for target 'msm8937_defconfig' failed
make[1]: *** [msm8937_defconfig] Error 1
Makefile:545: recipe for target 'msm8937_defconfig' failed
make: *** [msm8937_defconfig] Error 2
root@kali:~/Xiaomi_Kernel_OpenSource# 

`
I wanted to port Kali Nethunter to Redmi 3S and then add more drivers, but if i can't compile just the untouched kernel with the given instruction ([this](http://en.miui.com/thread-293285-1-1.html), but even there is no CONFIG_LOCALVERSION string in msm8937_defconfig ...) because something is wrong, and i don't know what, i slowly lose all hope on it.

What i have done?
-installed GCC from [this](https://github.com/offensive-security/kali-nethunter/wiki/Porting-Nethunter) tutorial
-downloaded the sources with `git clone -b land-m-oss https://github.com/MiCode/Xiaomi_Kernel_OpenSource.git`
-ran `make msm8937_defconfig` (found in arch\arm64\configs)
-ran make
-somewhere in this process i got some error, i tried everything but it still remained, deleted the sources, and started over

Yes, i know i don't need Kali Linux to build the kernel, but i'm low on storage and Kali was already installed in VMware.

So what is the problem? And please could you give me instructions how to compile just the untouched kernel?

luca020400, 2017-07-18:
If you aren't even able to follow a stupid guide and setup the environment ( like it's explained in the guide ), how do you plan to do anything with this kernel?

alamo18, 2017-07-18:
Lol noobs everywhere

MPeti1, 2017-07-19:
Sorry, i... i just use github very few times, and i totally forgot to search for the "Wiki" tab at the top of the page :D i used an other guide, which seems to be obsolete..
@luca020400 remembered me with mentioning "the guide". then i thought: "what guide? i.. i followed the guide, i also linked it.. ohhh there is a wiki section with THE guide :DDD"
sorry again, i just not use github frequently enough :D

closed 2017-07-21 by MPeti1